mavlink/qgroundcontrol #5796: **Feature Request** Camera trigger markers on screen

JonReacher, 2017-10-27:
It would be nice to have a small visual indication of camera triggers displayed on screen as they happen during a flight to have an indication of successful coverage and any gaps in coverage.

mhkabir, 2017-10-27:
This is already supported if you're running PX4 and your link supports Mavlink 2.0

DonLakeFlyer, 2017-10-27:
Not sure that needs mavlink 2.0? Either way, this already works in both PX4 and ArduPilot firmwares when running missions. Not sure what happens if you trigger the camera manually. But if it doesn't work then, it requires a firmware change to support it.

closed 2017-10-27 by DonLakeFlyer

JonReacher, 2017-10-27:
@DonLakeFlyer Is this supported in 3.2.4, or is it an option that needs to be activated? I'm not seeing it on any of my survey missions. Using PX4 with SiK telemetry radios.

DonLakeFlyer, 2017-10-27:
Supported in 3.2.4. Firmware should be sending on the following messages to QGC:
* MAVLINK_MSG_ID_CAMERA_FEEDBACK
* MAVLINK_MSG_ID_CAMERA_IMAGE_CAPTURED

One of those is PX4 and one is ArduPilot. Can't remember which is which. If you are getting any of those messages then something not happening on firmware side.

JonReacher, 2017-10-28:
@mhkabir Is this supported in PX4 1.6.5? Just so I know we're on the same page, I'm talking about a live graphic display of trigger events during a survey flight.

mhkabir, 2017-10-28:
Correct. The link needs to be running Mavlink 2 though, because our live feedback message is the mavlink 2 message space. Ping @bkueng @LorenMeier.

JonReacher, 2017-10-29:
How would I get the link to be running Mavlink 2?

bkueng, 2017-10-30:
@JonReacher It should be automatic, you might have to upgrade the firmware of the radio. You can force to use mavlink 2 by setting the `MAV_PROTO_VER` parameter to 2. On the shell, type `mavlink status`, and you see which protocol version is used.