mavlink/qgroundcontrol #5801: Cannot get gamepad on android to work

tve, 2017-10-28:
Running on android 7.1.2, using qgc v3.2.4, pixracer board w/PX4 connected via USB.
I have a gamepad paired, I can use it to navigate the android settings menu. I start QGC and have COM_RC_IN_MODE=0 (joystick/no rc checks). I restart QGC, no joystick menu shows in the settings.
Same gamepad works on Linux with QGC with same flight controller.
From what I can tell, this should work? How can I troubleshoot? Is there some debug log I can enable (console doesn't show anything I would associate with joystick).

DonLakeFlyer, 2017-10-30:
@gregd72002 Is there any logging to debug why a joystick may not be detected? Or is it just up to Qt to determine which ones it supports.

tve, 2017-11-02:
Is https://github.com/mavlink/qgroundcontrol/commit/e6e4dad915792590aa9cb69a24f1abd25245a0ef the culprit?

tve, 2017-11-02:
I tried v3.2.5 and it fixed the gamepad detection on Android!
However, I can't get it to calibrate. It detects the left stick down and up, but then shows the left stick right point for a fraction of a second and proceeds to show the left stick left point. There's a similar glitch for the right stick. Same gamepad calibration works fine on linux and the axis display tracks the gamepad motions just fine too.

DonLakeFlyer, 2017-11-02:
I think we are at the mercy of Qt's android gamepad support here.

DonLakeFlyer, 2018-06-20:
Not much we can do here since this comes from Qt. Some joysticks works, some don't.

closed 2018-06-20 by DonLakeFlyer