mavlink/qgroundcontrol #5814: QGC Waypoint protocol docs switch lat/long incorrectly

polastre, 2017-11-01:
On the qgroundcontrol page, it shows Param5 as Longitude and Param6 as Latitude.
http://qgroundcontrol.org/mavlink/waypoint_protocol

In both the QGC and Mission Planner code, Param 5 is **Latitude** and Param6 is **Longitude**. These should be switched in the docs.

QGC:
https://github.com/mavlink/qgroundcontrol/blob/master/src/MissionManager/MissionItem.cc#L413
https://github.com/mavlink/qgroundcontrol/blob/16541062efa0acee5e128f61dcd23106a7afc1f6/src/MissionManager/MissionItem.cc#L413

Mission Planner:
https://github.com/mairsbw/MissionPlanner/blob/master/GCSViews/FlightPlanner.cs#L1543

This is similar to the need for updated docs mentioned in #2243

DonLakeFlyer, 2017-11-01:
Anything from the old QGC web site is hopelessly out of date. @hamishwillee Should we delete these pages now?

closed 2017-11-01 by DonLakeFlyer

hamishwillee, 2017-11-01:
@DonLakeFlyer Not until you have reviewed them and assessed status. Even if they are hopelessly out of date we wouldn't necessarily just delete if there was still a need for a matching article in the new system.

In this case, specifically, does this document map to anything already on the new site? e.g. https://mavlink.io/en/protocol/mission.html . If so, is everything in the old document of relevance captured? If so, then yes, we could delete.

DonLakeFlyer, 2017-11-01:
> Not until you have reviewed them and assessed status.

I thought I already did all of that. Is there some other google doc I'm supposed to be looking at?

hamishwillee, 2017-11-01:
Not your fault, I said I would send you the review doc, but I see that I did not. Link now sent.