mavlink/qgroundcontrol #5816: firmware upgrade hangs on ubuntu 16.04

barchstien, 2017-11-01:
Hi Everyone

 - Working on Ubuntu 16.04
 - Using QGroundControl v3.2.5, freshly downloaded
   Tried both AppImage and Archive
 - Board is Pixhawk 2 (the "cube" stuff)
 - tried both ardupilot and px4 stable release, thought I don't think it matters at this stage

I start QGC, click on firmware, then it expects me to plug the flight controller to the usb port
Plugging, selecting the firmware, pressing ok, everything goes fine, until : 
"Programming new version..." hangs forever
Looking at the blue progress bar, it stops at random progress

I've done the same test on a windows machine, and it works fine (what a shame :-p )
I've, of course, removed modemmanager and added my user to the group : 
cat /etc/group | grep dialout
dialout:x:20:bastien

[image]

I can post more info if needed
Also fluent in c/c++/linux but never went into QGC code and would be glade to get hands on if someone points me to a specific place

Thanks in advance

barchstien, 2017-11-01:
Also, once the pixhawk 2 has been flashed (firmware installed) from windows, QGC on ubuntu 16.04 work perfectly fine... except for firmware install of course

DonLakeFlyer, 2017-11-02:
@dagar Do you have a Cube to test with?

barchstien, 2017-11-02:
yep, having it right before my eyes

DonLakeFlyer, 2017-11-02:
@barchstien Is it only the Cube your are having problem with. Or can you not flash any type of board?

barchstien, 2017-11-02:
@DonLakeFlyer only having this cube in my lab, other boards are in the main lab, that I gonna be visiting by the end of the month.
Do you know if I can get the output of the console when trying to upload the firmware ? That may give out some clues ?

DonLakeFlyer, 2017-11-02:
Turn on FirmwareUpgradeLog: https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html. Save the output and paste or zip here.

barchstien, 2017-11-02:
Here it goes
[5816_log.txt](https://github.com/mavlink/qgroundcontrol/files/1439687/5816_log.txt)
 - At he beginning : FirmwareUpgradeLog.debug=true\nFirmwareUpgradeVerboseLog.debug=true
 - Then I went into firmware upgrade, til it starts doing all the : _findBoardOnce
 - Then it finds the board, and downloads the available firmwares
 - Erase is fine
But nothing more, and QGC is frozen

barchstien, 2017-11-06:
ping ?

dagar, 2017-11-06:
I've been seeing this as well (QGC ~master built from source) on various pixhawks. It needs to be investigated and resolved before the next PX4 stable release.

dagar, 2017-11-06:
[qgc_px4fmu-v4pro_fail.txt](https://github.com/mavlink/qgroundcontrol/files/1446977/qgc_px4fmu-v4pro_fail.txt)

So far I can only confirm what @barchstien reported, but I don't see anything in the log.

dagar, 2017-11-06:
[image]

larsks, 2017-11-07:
Is this a dupe of #5519?

barchstien, 2017-11-07:
Ok, got some good news
I've forked/clone QGC repo
Compiled in debug and run/debug it, with some extra trace
And flashing the board started to work !
Typical race condition stuff, to my mind (works in debug or with extra trace, not in release)
So I added a 1msec sleep before the call to tcdrain(), and flashing works perfectly fine many times in a row. If I remove the 1msec sleep, flash hangs just like expected. Also using 10msec sleep helps, but flashing gets pretty slow. Using less than 1msec isn't enough to get flash to work
@dagar @DonLakeFlyer I guess this sleep is more a workaround than a fix. I could do a pull request if you want ?

Also, I've added : 
    `if (::tcdrain(fd) < 0) {
        qCDebug(FirmwareUpgradeLog) << "::tcdrain(fd) failed : " << ::explain_tcdrain(fd);
    }`
But tcdrain() just hangs, so I didn't get any more useful data

Could it be that the hanging is due to a bug in the UART driver ?
I'm curious about it, gonna try check it out, see if I get anything

barchstien, 2017-11-07:
Tried to find out which driver was used, using 
`dmesg | grep driver`
Gave back 
`[  130.123116] usbcore: registered new interface driver cdc_acm`
`[  130.123117] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters`
So it's cdc_acm driver
Looking [here](https://www.kernel.org/doc/Documentation/usb/acm.txt)
Sounds to me like a very mainstream, long existing driver, so I guess nothing wrong in there
Conclusion, I'm back to searching a way to fix that better than just using just before ::tcdrain()
`QThread::usleep(1000);`

DonLakeFlyer, 2017-11-07:
> Is this a dupe of #5519?

Most likely.

DonLakeFlyer, 2017-11-07:
Can you flash the board with APM Planner 2?

barchstien, 2017-11-07:
APM or PX4 doesn't make a difference
Flash hangs, except when adding msleep(1) before tcdrain()

DonLakeFlyer, 2017-11-07:
I'm talking about APM Planner2 the application.

barchstien, 2017-11-07:
ok, installing it

barchstien, 2017-11-07:
yes, flashing the firmware with APM Planner2 works fine

barchstien, 2017-11-08:
Been checking APM Planner2 code. It doesn't used termios (tcdrain), but rely on [QSerialPort](http://doc.qt.io/qt-5/qserialport.html)
Is it an option to move from termios to QSerialPort ?The former is cross platform, while termios is linux/unix only

DonLakeFlyer, 2017-11-09:
I can't seem to get 3.2.5 to hang, but master does. I didn't think it was possible to switch to QSerialPort since it didn't support direct writing. But if APM Planner to is using that then I guess it works.

DonLakeFlyer, 2017-11-10:
Linux folks can you build the pull and see if it works for you. I run Ubuntu in a VM and it can sometimes show different results than native. I think it's working now though.

barchstien, 2017-11-12:
@DonLakeFlyer Just cloned DonLakeFlyer:FirmwareUpgradeQSerialPort from https://github.com/mavlink/qgroundcontrol/pull/5835
Compiled it in Debug and Release and tried a couple of times to load the firmware... And it works like a charm !
Thanks
Still on Ubuntu 16.04 with Pixhawk 2

DonLakeFlyer, 2017-11-12:
Phew, good news. Thanks for testing.

aftersomemath, 2017-11-12:
I had the same issue on Ubuntu 16.04 and #5835 fixed the issue as well.

DonLakeFlyer, 2017-11-12:
Verified that is works on OSX, Ubuntu and Windows.

closed 2017-11-12 by DonLakeFlyer

vijay609, 2018-01-12:
I am hitting this in the latest version of the QGC 3.2.7 on Ubuntu 16.04. I am trying to install px4fmuv2_lpe version 1.6.3 (after downloading it locally and pointing QGC to it)

eirikaso, 2018-01-26:
I have this exact problem as well. 
Just downloaded the appimage again. Are these changes incorporated into the appimage file?

vijay609, 2018-01-26:
Here's what worked for me in QGC 3.2.7 on Ubuntu 16.04
1. Click the "Q" sign on the top left corner of the window.
2. Go to "General". In the "AutoConnect to the following devices:", uncheck every thing except for "Pixhawk".

DonLakeFlyer, 2018-01-29:
You need to use a daily build to get the fixes.

eirikaso, 2018-01-30:
Thanks!
Didn't work to uncheck all the boxes, but using the daily build did :)

dronecrewkari, 2018-02-02:
Actually, I have the same problem