mavlink/qgroundcontrol #5817: Multivehicle problem: RTK using too much bandwidth?

pythonmobile, 2017-11-01:
We are getting errors when we use RTK with multiple flying machines. One sample error is:

 Vehicle sent unexpected MISSION_ACK message error: Mission accepted (MAV_MISSION_ACCEPTED)

Without RTK, things seem to work. With RTK, even the parameter download slows down and times out. It seems that the RTK Streaming is using all the bandwidth from QGC. Is there a way to fix this/

We are using pixhawks + GCS (ubuntu) - both nightly build and stable have the problem + PX4 1.6.5.

If you want to reproduce this error, first connect RTK+USB (we use tiny from drotek), connect 3 machines to QGC over radio with 57600bps on the radios.  When you try to connect the 3 machines, you will start seeing parameter download errors already.

DonLakeFlyer, 2017-11-01:
Is each vehicle going through it's own seperate radio channel? You have three physical SiK radios connected to QGC?

pythonmobile, 2017-11-01:
@DonLakeFlyer Yes, indeed that is the system we have. All have unique IDs as well.

DonLakeFlyer, 2017-11-01:
Ok, then can you verify that this happens with a single vehicle as well. I don't se how multiple-vehicles makes any difference here.

pythonmobile, 2017-11-01:
@DonLakeFlyer We get this error when we get to three vehicles. One vehicle works fine. 2 is sluggish. 3 starts to time out. 4 is time out. With 4 Vehicles without RTK - is sluggish but works. With RTK - fails to download parameters.

DonLakeFlyer, 2017-11-01:
@bkueng Could you double check the code to make sure that somehow the RTK stream isn't getting doubled/tripled when multi-vehicle. Doesn't make sense that multi-vehicle would affect this.

pythonmobile, 2017-11-01:
@bkueng @DonLakeFlyer  Is there a way to control how much bandwidth RTK updates are using?

bkueng, 2017-11-03:
I checked the code, and it looks ok to me: each RTCM message is sent to each connected vehicle. Sounds to me this is either a hardware setup problem (radio interference for instance), or a more fundamental problem with multi-vehicles, which is revealed by RTK because it needs more bandwidth.

@DonLakeFlyer is there a way to inspect the current bandwidth on the lowest level for each link (both RX and TX)? 
@pythonmobile can you do the following: with RTK connected & running, check and provide the output of `mavlink status` on the shell for each connected vehicle,
- for only a single connected vehicle
- for 3 connected vehicles

> @bkueng @DonLakeFlyer Is there a way to control how much bandwidth RTK updates are using?

No, this is pretty much given, and depends on the number of sattelites you see.

DonLakeFlyer, 2017-11-04:
> is there a way to inspect the current bandwidth on the lowest level for each link (both RX and TX)?

No there isn't. I"m trying to figure out some way to debug this from the QGC side. Does mavlink status tell you anything about bandwidth. I"m wondering if all mavlink messages are being sent to all vehicles due to some strange hardware problem.

Do you only get multi-vehicle communication problems like parameter issues when RTK is connected?

bkueng, 2017-11-06:
> Does mavlink status tell you anything about bandwidth

yes exactly: both RX and TX

DonLakeFlyer, 2018-01-04:
@pythonmobile Is this still causing problems for you?

DonLakeFlyer, 2018-06-18:
Closing due to no response

closed 2018-06-18 by DonLakeFlyer

paan2097, 2018-06-27:
@pythonmobile @DonLakeFlyer @bkueng 

Hi Guys! I am using px4 with the Here+ on the plane and the base connected to QGroundControl, works great. I am looking to buy a second Here+ because I do multi vehicle flights as well, do I just need one base RTK module and QGC will send the messages to all connected vehicles? Or do I need multiple bases?

hamishwillee, 2018-06-28:
@paan2097 Don't hijack existing threads please. This is a support question, and should go in discussion boards not issues.
In short though, you should only require one base station.