mavlink/qgroundcontrol #5819: QGC daily build: Mission transfer failed.

Dnrodriguez374, 2017-11-02:
Mission transfer failed. Retry transfer. Error: Vehicle did not request all items from ground station. MISSION_REQUEST

I can connect easily to my vehicle, but I cannot read/load/upload way-points. I can only clear them. 

This only happens with the daily build, with the stable build this error does not come up. 

I am using a pixhawk with firmware 1.6.5 stable and using QGC version 3.2.5 stable and the latest daily build.

DonLakeFlyer, 2017-11-02:
Firmware 1.6.5 has a mavlink bug which prevents the mission protocol from working correctly. This has been fixed in master firmware.

closed 2017-11-02 by DonLakeFlyer

sotiris155, 2018-03-01:
Does the problem still exist?
I get same message on Windows, Ubuntu and Android 7.0 (different devices obviously). 
QGC 3.3.0

DonLakeFlyer, 2018-03-13:
Are you using the latest firmware? What version?

sotiris155, 2018-03-13:
1.6.5 not uploading missions
1.7.3 seems to have more problems, not calibrating etc

I'm using mission planner currently with ArduPilot firmware 3.5.3, no problems here.

DonLakeFlyer, 2018-03-13:
As stated above, 1.6.5 does not work with mission transfer. If you are having calibration problems with 1.7.3 then you should enter a separate issue for that.