mavlink/qgroundcontrol #5826: Changing speed and altitude during a mission

fredrikfalkman, 2017-11-06:
It would sometimes be useful to be able to change speed or altitude during a mission or a guided go-to.

The question then arrises: For how long? Until the next waypoint, relative for the rest of the mission or absolute for the rest of the mission? I don’t know what would be best - possibly something the user could could choose.

In the same category it might be useful to have an evasive manoeuvre - usually a hard starboard?

What this is really about is to bring in a little more of the remote pilot in the autonomous modes for those that can not fall back on RC control.

DonLakeFlyer, 2017-11-07:
@antiheavy FYI...

Anothing thing I had in the back of my head was for fixed wing landings to nudge left/right or up/down in case it's not nailing the landing spot you want.

Antiheavy, 2017-11-07:
> Anothing thing I had in the back of my head was for fixed wing landings to nudge left/right or up/down in case it's not nailing the landing spot you want.

YES!  I've had this on my list of items for a long time and was planning to make a feature request this week in fact.

> What this is really about is to bring in a little more of the remote pilot in the autonomous modes for those that can not fall back on RC control.

I really like this idea.  "Altitude mode" doesn't quite fit the bill in my experience.  Would be great to find something smarter.

Big topic so I'll humbly submit my thoughts:

One question is how much of this could be handled only by QGC and how much requires new firmware support.

"Bump Landing" is my immediate interest so I'll start here.   There are a couple approaches: do you try to update the Land waypoint position or do you try to send some guidance input directly (like "joystick" commands) that the flight controller interprets.  I believe the multicopter folks have some sort of joystick override in auto-landing, this might be something to leverage for fixed wing too.  The details of how it gets implemented are important (i've been through this exact topic on a number of systems); it is important to limit how much "bump" is allowed.

"Airspeed and Altitude override":  In most past systems I've worked with we've had some form of this.  My favorite UI device for this is that the altitude and airspeed displays are actually buttons that when pressed some dialog pops up nearby to toggle on/off and set a value.  Another approach that works well for airspeed is to have only 2 set values: "cruise" (or "auto") and "dash" - having fine airspeed control usually isn't required and it is similar to just have the two options.  Though is needed for how and when the system would automatically kick you out of override.

"Evasive Maneuver" I've never used a system that had something like this, although I've heard of systems that do.  I think that some of the ideas for improving guided mode in this issue https://github.com/mavlink/qgroundcontrol/issues/5748 to more easily "Pause" and change location and altitude might be enough to satisfy me.  I'm interested to hear ideas how this might be implemented:  Hard right bank for X seconds, or 90 degree heading change for X seconds, other ideas?

Great topic.

fredrikfalkman, 2017-11-07:
> My favorite UI device for this is that the altitude and airspeed displays are actually buttons that when pressed some dialog pops up nearby to toggle on/off and set a value.

I like that. Might I point to iOS 11's new control center widgets as a quick way to change settings:

https://cdn.cultofmac.com/wp-content/uploads/2017/07/ios-11-timer.gif

> Another approach that works well for airspeed is to have only 2 set values: "cruise" (or "auto") and "dash"

Also a good point. I would sugest:

VBE | Best endurance speed – the speed that gives the greatest airborne time for fuel consumed
VBR | Best range speed – the speed that gives the greatest range for fuel consumed 
VNE | Never exceed speed - OR - Dash

anandbaskaran, 2018-03-09:
Hello, 

It could be a cool feature of qGroundControl that I am eager to try it out. 

However, does PX4 already supports this feature? If so, can you please let me know the mavlink commands that can be used?

closed 2024-06-01 by HTRamsey

anandbhaskaran, 2024-06-01:
Great work @HTRamsey 🚀