mavlink/qgroundcontrol #5832: Switch Thresholds not respected

tops4u, 2017-11-10:
Running QGroundcontrol Master (about 2 Weeks old from today), running on MacOS HighSierra with AUAV-X2 connected via USB running on current Master. 

I wanted to change the Threshold for the Kill-Switch. My Remote Killswitch basically has two positions but if the switch is pushed just about halfway there is in fact a third position, which would already trigger the Kill switch. In order to make sure Kill Command is only issued if the Switch is completely flipped, I wanted to adjust the Threshold of the Switch (Default 0.25), to check if the value is correct I looked at the FlightMode Tab in QGC. After trying various values I noticed that the behavior never changed. So I checked with the MAVLink Messages. Even if using ridiculous values for the Threshold (like 2) that the Flight Controller never triggered the Kill, QGC shows an active KillSwitch even when reaching 1500us PWM witdth. I assume the Threshold is hardcoded but should assume the Value of RC_KILLSWITCH_TH.

DonLakeFlyer, 2017-11-10:
You seem to have two problems here:
1) The firmware is not respecting your RC_KILLSWITCH_TH value. That would be a firmware bug.
2) RC_KILLSWITCH_TH must be something that got added but nobody ever told me about. QGC currently knows nothing about this so it displays ui based triggered >= 1500 PWM. So that needs to be fixed on QGC side. But that is just a visual indicator and has no affect on whether the kill switch actuall works or not.

tops4u, 2017-11-10:
@DonLakeFlyer Just the later is the case. The visual representation of an active KillSwitch in the Flight Modes Panel does not correspond to what the Flight Controller does. 

Firmware is handling the Threshold correctly.

DonLakeFlyer, 2018-06-20:
This was pretty broken. For now I have to disconnect the highlighting and I will work on adding back next release.

thomasgubler, 2018-06-20:
➤ Don Gagne commented:

This was pretty broken. For now I have to disconnect the highlighting and I will work on adding back next release.