mavlink/qgroundcontrol #5833: At initial Connection Mode is not updated

tops4u, 2017-11-10:
Running QGroundcontrol Master (about 2 Weeks old from today), running on MacOS HighSierra with AUAV-X2 connected via USB running on current Master.

When Booting Up and RC TX is already on and Flightmode is set to PosCtl for example. This is not reflected in the Top Bar:

[image]

In this case, the FlightMode Switch was in PosCtl Mode when Powering up. FC denied PosCTL as there are checks failing (like missing GPS) in reality the FC fell back to Alt Hold Mode, while the Status Bar showed Stabilized.

DonLakeFlyer, 2017-11-10:
QGC is just displaying whatever it gets back in the HEARTBEAT message for flight mode. WHen you are in this state where you think QGC is displaying something wrong can you open the mavink inspector and take a screen shot of the HEARTBEAT message so I can check which is coming back from there.

tops4u, 2017-11-17:
I think this may be a Firmware Issue. It works fine if TX is switched on after the Aircraft, but if it is the other way round Stabilized or Manual Mode is assumed. Shall I open an Issue on the Firmware Side?

Situation when Aircraft is powered after TX was on and FlightMode PosCtrl was selected:
[image]

Situation after toggling Flight Mode Switch -> Stabilized & back -> PosCTL
[image]

DonLakeFlyer, 2017-11-17:
Yup. Nothing QGC can do about it.

closed 2017-11-17 by DonLakeFlyer