mavlink/qgroundcontrol #5840: Change the MAVLink #WIND message output to 0-360

fredrikfalkman, 2017-11-14:
I think that it would make more sense for most pilots to get the wind direction in 0-360 degrees rather than the current -/+ 180 degrees from True North that is the result of the ArduPlane implementation:

https://github.com/ArduPilot/ardupilot/blob/8178ab4037b05c515946f1f3fe501babaf48ad5b/ArduPlane/GCS_Mavlink.cpp#L296-L305

Noticing that MAVLink spec doesn't prescribe how to present wind direction:

http://mavlink.org/messages/ardupilotmega#WIND

Referencing this discussion: https://github.com/ArduPilot/ardupilot/issues/7221

DonLakeFlyer, 2018-03-23:
Done

fredrikfalkman, 2018-03-23:
Great! Will try soon!

DonLakeFlyer, 2018-07-18:
Get a telemetry log for the flight as well.

Kjkinney, 2018-07-18:
Here's a telemetry log from yesterday, forgot the QGroundControl auto-saved those. The wind wasn't very strong, but the local weather station was saying about 4kts from 20degrees. I'll look for some more with higher winds.
[2018-7-18 Telemetry Logs.zip](https://github.com/mavlink/qgroundcontrol/files/2206467/2018-7-18.Telemetry.Logs.zip)

Kjkinney, 2018-07-18:
Wind on this day was from the southeast at around 10 mph
[2018-7-11 Telemetry Logs.zip](https://github.com/mavlink/qgroundcontrol/files/2206532/2018-7-11.Telemetry.Logs.zip)

closed 2018-07-31 by DonLakeFlyer