mavlink/qgroundcontrol #5843: last github version of QGC fails to compile

john9393, 2017-11-16:
# **Howdy all.**
**_Last GITGUB version of QGC fails to compile on Linux-64_**

**gcc --version : gcc (GCC) 7.2.0**
**qt5 version : 5.9.0**

## result of qmake :
 Project MESSAGE: Qt version 5.9.0
Project MESSAGE: Linux build
/usr/bin/ccache
Project MESSAGE: Found ccache, enabling
Project MESSAGE: QGroundControl Development master:909dd828a 2017-11-16 05:20:28 +0000
Project MESSAGE: Release flavor
Project MESSAGE: Using Default QtLocation headers
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
Project MESSAGE: Including support for video streaming

## and make fails with :
compiling ../src/MissionManager/QGCMapPolygon.cc
In file included from ../src/MissionManager/QGCMapPolygon.cc:11:0:
../src/QGCGeo.h:25:24: error: expected unqualified-id before « { » token
 #define sincos(th,x,y) { (*(x))=sin(th); (*(y))=cos(th); }
                        ^
In file included from /usr/include/features.h:423:0,
                 from /usr/include/c++/7.2.0/x86_64-slackware-linux/bits/os_defines.h:39,
                 from /usr/include/c++/7.2.0/x86_64-slackware-linux/bits/c++config.h:533,
                 from /usr/include/c++/7.2.0/type_traits:38,
                 from /usr/include/qt5/QtCore/qglobal.h:45,
                 from /usr/include/qt5/QtCore/qnamespace.h:43,
                 from /usr/include/qt5/QtCore/qobjectdefs.h:48,
                 from /usr/include/qt5/QtCore/qobject.h:46,
                 from /usr/include/qt5/QtCore/QObject:1,
                 from ../src/MissionManager/QGCMapPolygon.h:13,
                 from ../src/MissionManager/QGCMapPolygon.cc:10:
/usr/include/bits/mathcalls.h:79:1: error: expected unqualified-id before « throw »
 __MATHDECL_VEC (void,sincos,,
 ^
make: *** [Makefile:50424: obj/QGCMapPolygon.o] Error 1compiling ../src/MissionManager/QGCMapPolygon.cc
In file included from ../src/MissionManager/QGCMapPolygon.cc:11:0:
../src/QGCGeo.h:25:24: error: expected unqualified-id before « { » token
 #define sincos(th,x,y) { (*(x))=sin(th); (*(y))=cos(th); }
                        ^
In file included from /usr/include/features.h:423:0,
                 from /usr/include/c++/7.2.0/x86_64-slackware-linux/bits/os_defines.h:39,
                 from /usr/include/c++/7.2.0/x86_64-slackware-linux/bits/c++config.h:533,
                 from /usr/include/c++/7.2.0/type_traits:38,
                 from /usr/include/qt5/QtCore/qglobal.h:45,
                 from /usr/include/qt5/QtCore/qnamespace.h:43,
                 from /usr/include/qt5/QtCore/qobjectdefs.h:48,
                 from /usr/include/qt5/QtCore/qobject.h:46,
                 from /usr/include/qt5/QtCore/QObject:1,
                 from ../src/MissionManager/QGCMapPolygon.h:13,
                 from ../src/MissionManager/QGCMapPolygon.cc:10:
/usr/include/bits/mathcalls.h:79:1: error: expected unqualified-id before « throw »
 __MATHDECL_VEC (void,sincos,,
 ^
make: *** [Makefile:50424: obj/QGCMapPolygon.o] Error 1

DonLakeFlyer, 2017-11-17:
My guess is problem with newer gcc. Ubuntu 16.04 works fine (as do travis CI builds) with the version it has which seems to be 5.4.

dogmaphobic, 2017-12-01:
Fixed in #5854

closed 2017-12-01 by dogmaphobic