mavlink/qgroundcontrol #5850: Feature request: Survey / Lock copter orientation to Grid Angle

ThorstenDP, 2017-11-19:
Hi all,

Tower has an important feature for mission planning, which is missing in QGC so far:
The option to set the WP_YAW_BEHAVIOR = 0 ("Never change yaw") when the Copter has reached the first WP. 

There are several advantages:

- Security: you always know the orientation of the copter in case you have to fly manually
- Easy image analysis: scanning through the images visually it is much intuitive if they all show the same orientation
- Saving battery and time: It should be (at least a little bit) more efficient if any additional movement is avoided.

The main objective however is security. Knowing the orientation is especially important when the mission covers a larger area and you operate at the limits of line of sight. If the copter is 300+ m away it is very hard to see if it turns, how fast it is and if it is still flying away from you or already coming back to you. Keeping track of the copter is crucial. If you become distracted for just a second it is sometimes hard to find it again - not to mention its flight direction. It also strongly depends on your position relative to the survey grid, as well as the distance between you and the copter. This is for sure no problem for small area missions, but it is for any grid larger than 250*250m. If yaw is fixed you always know the orientation of the copter. This is why I never change it. It might be a life saver. You can do it manually by positioning the copter alnong the grid survey lines, but this is never prefect.

Maybe the easiest way implementing it would be:

    take off
    go to WP1
    turn the copter towards the 2nd WP
    set WP_YAW_BEHAVOR = 0
    proceed mission
    RTL

The implementation in Tower seems to be a little more complex. However it works perfect. I would really appreciate to see this feature in QGC as well.

Thanks and cheers,
Thorsten

PS: There is a longer discussion at [https://github.com/DroidPlanner/Tower/issues/1473](https://github.com/DroidPlanner/Tower/issues/1473) and other isses linked there

ThorstenDP, 2018-01-11:
Would be great to see this in v3.4.

DonLakeFlyer, 2018-01-12:
It seems very strange that a mission would set a parameter. Also changing the orientation of the vehicle will screw up survey image calculations. The map view will show you where the vehicle is and orientaion in relation to home position. Why doesn't that work?

ThorstenDP, 2018-01-12:
Sure the map view helps. But there are three scenarios where it might where the fixed yaw is beneficial:

Telemetry loss

- If the telemetry lost connection, the map view does not help. I have this problem frequently due to the regulations of the transmission power of the radio.

Panik

- in case of some unexpected event (birds, planes, helicopters crossing, etc.) you have to react quickly. trying to orientate on the map might be too time consuming.

Distraction

- even if you are distracted by something or lost the LOS to the drone, you have a chance to bring is back quickly if you know the orientation relative to you.

Hence, I was really happy when this feature was implemented in Tower. Actually, it is the only reason why I am still using it sometimes. 
What I did before and what I am doing now with QGC is to set WP_YAW_BEHAVIOR to 0, position the copter as good as possible along the flight path and then start the mission. This works somehow but you can easily end up with 5-10° offsets. 
The way it is implemented in Tower is by setting MAV_CMD_CONDITION_YAW after each waypoint. Not sure if itis required after each WP. I am wondering if it is possible to set WP_YAW_BEHAVIOR to 0 when planning the mission in the same way as WP_NAV_SPEED is set. For sure in this case one "override" of the current yaw has to happen somewhen.

closed 2026-10-04 by github-actions[bot]