mavlink/qgroundcontrol #5852: Mission transfer failed / Refly at 90 degree offset

ThorstenDP, 2017-11-20:
Hi all,

mission upload fails when "Refly at 90 degree offset" is checked.
```
Received mission iten out of sequence
(MAV_MISSION_INVALID_SEQUENCE). Sequence: 1.
```

Best regards,
Thorsten

ThorstenDP, 2017-11-20:
It seems the limit is 718, which was introduced in AC3.2: 
[https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/ReleaseNotes.txt#L751](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/ReleaseNotes.txt#L751)
So it is not an QGC issue. However, a warning would be nice. I'll open a request to extend the number of possible waypoints at [https://github.com/ArduPilot/ardupilot](https://github.com/ArduPilot/ardupilot)

ThorstenDP, 2017-11-20:
At https://github.com/ArduPilot/ardupilot/issues/7254#issuecomment-345784356 @WickedShell replied that:
> MAV_MISSION_INVALID_SEQUENCE is only sent when a GCS uploads a waypoint that the autopilot was not requesting. This is a GCS mission handling error...

WickedShell, 2017-11-20:
@ThorstenDP the tlog from this will help whoever gets to doing the review of the issue as it will have the record of sent/received messages and allow diagnosing why the upload failed.

ThorstenDP, 2017-11-21:
It seems to be related to size. I made several tests and as soon as the number of WPs is too high I receive the 'out of sequence' error. I'll prepare a tlog anyway. The tests were made with a normal SiK Telemetry Radio V2 433Mhz from mRo and default settings.

DonLakeFlyer, 2017-12-09:
It's going to be next to impossible to warn correctly about a waypoint limit. Since the limit varies by firmware type and firmware version. What is really needed is an actual "too many waypoints" error code in the mavlink spec. Then the firmware could send that and ground station chould show something useful.

hamishwillee, 2020-04-26:
This can now be fixed - MAVLink allows firmware to report [MAV_MISSION_TOO_MANY_ITEMS](https://mavlink.io/en/messages/common.html#MAV_MISSION_TOO_MANY_ITEMS)

closed 2026-10-04 by github-actions[bot]