mavlink/qgroundcontrol #5866: Feature request: Survey missions option to first fly all even lines then fly all odd lines on the way back

deonblaauw, 2017-12-04:
We fly missions using a fixed-wing drone, and would really find it useful if the survey missions could be setup such that the drone flies all the even paths on the first pass, then all the odd paths on the way back. Doing this means that the aircraft would be able to settle back onto the path quicker, making for more efficient missions (currently we lose some image quality on the mission edges due to "back to the path" settling time). This is especially relevant when high image overlap is required which leads to tightly spaced lines.

sanderux, 2017-12-04:
And the vehicle would be back where it started when it finishes the mission.
@DonLakeFlyer @dogmaphobic

JonReacher, 2017-12-07:
I agree. Especially useful in areas sensitive to flying extended turnarounds over nearby property/ airspace.

DonLakeFlyer, 2017-12-09:
Using a turnaround distance doesn't fix the problem of cleaning up the pat to entera transect ?

sanderux, 2017-12-09:
That could almost double the mission distance

DonLakeFlyer, 2017-12-09:
Sure, but don't you still need to use some amount of turnaround anyway? Without turnaround fixed-wing entry will almost always be off. Seems like this would only be needed if the FW min turning radius is larger than the transect spacing? Does that happen often?

Not agruing against supporting this. I want to make sure I understand various FW survey problems correctly.

sanderux, 2017-12-09:
The turn radius on our vehicle is 100m, doing a half circle it will be fully level at the end. Generally the line distance is between 20 and 50 meters, so even 3 or 4 passes would be nice to have.

sanderux, 2017-12-09:
If you set max bank angle in sitl to about 30 degrees you will see what happens

JonReacher, 2017-12-10:
@DonLakeFlyer To answer your question I'll give you a typical scenario for myself. A transect width of 200' wide is common for a 400' altitude survey @ 70% side lap w/ an APS-C sized sensor. Very common.
 Our AV takes about a 3-400' wide turn in the turnarounds, which means 3-400' long turnarounds just to be level by the time we hit the new transect. Eats up a lot of mission time. An odd/ even arrangement would be perfect for doing a large reduction in turnaround space needed.
Sander's comment about limiting bank is probably pretty common as well. We have a limit set.

DonLakeFlyer, 2017-12-11:
Got it. Thanks everyone for the detailed information. Want to make sure I'm educated on all this stuff.

Antiheavy, 2017-12-23:
+1 for a feature like this.  Very useful for certain fixed wing survey applications.  For those of us who live in the frozen north, we call this a zamboni pattern.

ThorstenDP, 2018-01-11:
+1 not only for fixed wing surveys. This might also have positive effects for generating orthomosaics. Some discussion at: https://github.com/DroidPlanner/Tower/issues/1474

Antiheavy, 2018-06-02:
check out recent QGC daily builds.  We flight tested this yesterday on a fixed wing and it works pretty awesome.

DonLakeFlyer, 2018-06-02:
Ah, I new there was an issue on this somewhere. So  this is implemented now. What is not is discussion on how to handle the turnaround between between the two passes. That is discussed in #6451

closed 2018-06-02 by DonLakeFlyer