mavlink/qgroundcontrol #5870: Solo video won't work with updated solos.

AndKe, 2017-12-07:
Currently, we check for Solo by looking for STATUSTEXT "APM:Copter solo-*"
if it's found, we do the connectToHost(_artooIP, _artooVideoHandshakePort);
which would activate the Solo video..   
More and more Solos are running ArduCopter code, and they er not that easy to identify.
... and it's kind of dirty to expect user to run "nc 10.1.1.1 5502"  to make it work.

I see two good solutions:
1: try the connection blindly, it does not harm as long it does not generate lots of traffic (which it dont)
2: have an checkbox, an option to try this (and/or maybe other ip/port) connection.

BigOkie, 2018-05-05:
I believe this is happening to me as well; running a Solo with Arducopter 3.5.4 firmware.  Tower displays video just fine, QGC not so much.

dagar, 2018-05-05:
@BigOkie can you expand on that? Does QGC video sometimes work, but intermittently?

DonLakeFlyer, 2018-05-05:
The Solo needs to be told from the ground station to start video. The trigger for QGC to do that is based on text message output which doesn't happen when you are using an ArduPilot build on the solo. Only for original Solo firmware. I just haven't had time to get around fixing this yet.

> in case of solution 1 , I could test and submit a PR with that...

@AndKe If you could make the pull that would be great.

AndKe, 2018-05-05:
https://github.com/mavlink/qgroundcontrol/pull/6425

I know that the IMX6 from solo have been integrated into some lager RPAS, so if we se absolutely no side effects, this approach may very well be expanded to any vehicle.

- testet, works fine for OpenSolo + ArduCopter  (not only the old Solo-fork)

BigOkie, 2018-05-08:
@dagar I used to get it to work intermittently but thinking back on that now it was likely due to me still having the Solo firmware loaded, and not Arducopter.  I'll just wait for the correction.  I have Solex also and while it seemed to have given me fits in the past, it seems to be working much better now.  Not sure why.

closed 2018-05-08 by DonLakeFlyer