mavlink/qgroundcontrol #5872: Add gimbal control to fly view

DonLakeFlyer, 2017-12-10:
From OldGazer on RC Groups:
If I power up the aircraft and connect with MP and go to the Actions tab I can retract the mount, set it to MavLink targeting, set it to RC Targeting, and I if I right click on the map I can "point and click" (Point Camera Here), or use GPS coordinates to point the camera.    

I have not found any of that functionality in APM Planner.  I did find "point camera here" in qGC, and the modes (MavLink, RC, etc) are "buried".

The point is, in Mission Planner I can change from MavLink to RC targeting and back again, I can retract the mount, I can put the mount in Neutral, all from the Actions tab in the Flight Data screen.  Neither APM Planner or qGround Control have that functionality on there equivalent screens.

dogmaphobic, 2017-12-10:
@sanderux is looking for related functionality as well. His input could better shape this.

sanderux, 2017-12-18:
So there are several controls that would be needed:
- Pan
- Tilt
- Zoom
- Video switch
- Retract
- Function switch (should be configurable how many positions and ideally nameable) For example: return to neutral position

Channels (commonly the aux channels) should be assigned for every function.
The functions should be controllable from the video tab. 
For Pan, Tilt and zoom it should be controllable if it is in position or velocity mode (position would slowly increase/decrease the value, velocity would set it high/low while pressed, and return to neutral when released)

DonLakeFlyer, 2017-12-31:
Most of this is already covered by ArduPilot mavlink messages which we could bring across to common.

DonLakeFlyer, 2017-12-31:
Related to #3920

DonLakeFlyer, 2017-12-31:
Sander's comment seems to cover two areas:
1) Configuring the firmware so the gimbal can be controlled from a transmitter (I think?)
2) Controlling the gimbal from UI commands (and missions as well)

I won't comment on #1 since that is more of a firmware setup thing as opposed to ground station. 

For #2, a far as I can tell everything is covered by the existing MAV_CMD_DO_MOUNT_CONFIGURE and MAV_CMD_DO_MOUNT_CONTROL commands. And poking through PX4 Firmware code seems like a bunch of it is implemented already.

The thing missing is getting gimbal status back to the ground station so you can show current gimbal angles in some sort of display. I think ArduPilot already has something like that in their dialect so would be good to ask them about it and if so, carry across their stuff to common is possible.

LorenzMeier, 2017-12-31:
We can't carry dialects across as this conflicts with other people's dialects. These should never have gone into a dialect in the first place.

DonLakeFlyer, 2018-06-20:
@dogmaphobic Did you add this?

dogmaphobic, 2018-06-20:
Found the PR: https://github.com/mavlink/qgroundcontrol/pull/6138

closed 2018-06-20 by DonLakeFlyer

RyanHurst, 2019-07-10:
Why was this issue closed if it wasn't ever merged?

dogmaphobic, 2019-07-11:
There is work being done to this. A new PR is coming shortly.

Mohit953, 2021-07-17:
Is there any update on this issue ?