mavlink/qgroundcontrol #5876: 6s delay before GUI is displayed on QGC start.

AndKe, 2017-12-11:
current master, deleted(and created new) QGroundControl.ini
on start I see: 
```
 ./qgroundcontrol-start.sh 
inotify_add_watch("/home/andre/.config/ibus/bus/0ec43c14e6153f2da3b24205507c77d2-unix-0") failed: "No space left on device"
No text-to-speech plug-ins were found.
```

then it takes 6-7seconds before GUI suddenly spawns.

The system is not slow, APMPlanner2 spawns it's GUI in ~2 sec (I assume it's splashscreen that appears instantly is the delay)

DonLakeFlyer, 2017-12-11:
Same thing for AppImage?

AndKe, 2017-12-11:
where do I get that ?

AndKe, 2017-12-11:
ok, found at https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html  - will test

AndKe, 2017-12-11:
Appimage starts ~2s after text output, faster. - it does not complain/contain speech.  - maybe that's the difference ?

DonLakeFlyer, 2017-12-11:
My quess is that Qt is searching high and low through your harddisk for a plugin which is causing the slow down. This will go away once we move to new Qt version which fixes the speech thing.

AndKe, 2017-12-11:
just discovered that Qt5.9.2 is required, downloading on slow hotel wifi now, but it sounds like It's not going to fix speech, which version does fix it ?

DonLakeFlyer, 2017-12-11:
Why not just use AppImage? Not sure which new Qt is needed for speech. @dogmaphobic would know.

AndKe, 2017-12-11:
I like to try out the new features. :)

DonLakeFlyer, 2017-12-11:
There are Daily Build AppImages available as well.

AndKe, 2017-12-11:
I see now :)  - will figure out why mine takes time... Thanks.

closed 2017-12-11 by AndKe