mavlink/qgroundcontrol #5889: RTCMMavlink error?

RoelHelsen86, 2017-12-15:
I was browsing the code for RTK handling and I think I found an implementation error in RTCMMavlink.cc. It would be nice if someone from the developers could take a look at it and see whether it is an actual bug, or that I misinterpreted the code.

In the code, it increments both the sequence Id and the fragment Id for every fragment that is sent. However, ArduCopter expects different fragments to have the same sequence Id, otherwise it will remove the previously received fragments and starts a new buffer.

qgroundcontrol:

    if (message.size() < maxMessageLength) {
        mavlinkRtcmData.len = message.size();
        memcpy(&mavlinkRtcmData.data, message.data(), message.size());
        sendMessageToVehicle(mavlinkRtcmData);
    } else {
        // We need to fragment

        static uint8_t sequenceId = 0;  // Sequence id is used to indicate that the individual fragements belong to the same set
        uint8_t fragmentId = 0;         // Fragment id indicates the fragment within a set

        int start = 0;
        while (start < message.size()) {
            int length = std::min(message.size() - start, maxMessageLength);
            mavlinkRtcmData.flags = 1;                      // LSB set indicates message is fragmented
            mavlinkRtcmData.flags |= fragmentId++ << 1;     // Next 2 bits are fragment id
            mavlinkRtcmData.flags |= sequenceId++ << 3;     // Next 5 bits are sequence id
            mavlinkRtcmData.len = length;
            memcpy(&mavlinkRtcmData.data, message.data() + start, length);
            sendMessageToVehicle(mavlinkRtcmData);
            start += length;
        }
        if (sequenceId == 0x1F) {
            sequenceId = 0;
        }
    }

ArduCopter:

    if (rtcm_buffer->fragments_received &&
        (rtcm_buffer->sequence != sequence ||
        (rtcm_buffer->fragments_received & (1U<<fragment)))) {
        // we have one or more partial fragments already received
        // which conflict with the new fragment, discard previous fragments
        memset(rtcm_buffer, 0, sizeof(*rtcm_buffer));
    }

In my opinion, the sequence Id should be kept outside the if statement and only incremented and checked afterwards.

Unfortunately I cannot provide any data to show that this indeed is a problem, but I just want raise awareness of this (potential) problem.

DonLakeFlyer, 2017-12-16:
@bkueng Can you help here.

bkueng, 2017-12-18:
PX4 does not use the sequence & fragment ID's, these were added later, specifically for APM as it seems.

From the message definition in http://mavlink.org/messages/common#GPS_RTCM_DATA, it is not entirely clear to me when the sequence id is supposed to be incremented. This should be clarified.
It would make sense to increment it only for new messages and not within the fragments. Which is in agreement with @RoelHelsen86 , but it's rather a question of how we want to define the message field.

RoelHelsen86, 2017-12-18:
This should probably be discussed with the developpers of ArduPilot (which I'm not) 
If px4 does not use the ID's at all, it might be a good idea to put the increment outside the while-loop, and preferably outside the if-statement such that the sequence ID is only incremented for each message and that the fragment ID will make the message ID unique. In that case it will work for both ecosystems.

DonLakeFlyer, 2017-12-18:
@tridge Can you help as to how this needs to work with ArduPilot? Trying to get something to work with both PX4 and ArduPilot.

DonLakeFlyer, 2017-12-20:
Talked with @oxinarf on the ArduPilot. What @RoelHelsen86 is correct:
* Only fragment id should increment for each fragment
* Sequence id should increment to indicate a new set. Single fragment or multiple fragment.

@bkueng if PX4 doesn't use these how does it know if a set of fragments is complete?

bkueng, 2017-12-20:
> @bkueng if PX4 doesn't use these how does it know if a set of fragments is complete?

It doesn't care :) It let's the GPS device handle (partial) drops. And if it's just fragmented, it's ok to only feed these one at a time (there are no other writes to the GPS device).

DonLakeFlyer, 2017-12-20:
MAVProxy implementation as example: https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_DGPS.py#L26-L71

DonLakeFlyer, 2017-12-20:
@bkueng Could you change this code to match how MAVProxy does it?

RoelHelsen86, 2017-12-21:
On a side note, is there a reason why qgroundcontrol uses RTCM 1005, 1077, 1087 and Mission Planner uses 1005, 10074, 1084 and 1230?
(I know they are MSM4 vs MSM7, but I don't know what the difference between them is)

What is the benefit of one over the other and shouldn't qgroundcontrol also include the 1230 message for glonass?

RoelHelsen86, 2017-12-21:
PS: In normal operation, the problem I discovered is not a big one. Typically the length of the RTCM messages falls under the 180 limit and the first case applies which works for both arducopter and px4

bkueng, 2017-12-23:
> What is the benefit of one over the other and shouldn't qgroundcontrol also include the 1230 message for glonass?

QGC uses the messages that were recommended from ublox support. MSM4 has a bit reduced bandwidth requirements, but also reduced accuracy (I don't know how much the resulting precision is affected).

The u-blox M8P [docs](https://www.u-blox.com/sites/default/files/NEO-M8P_DataSheet_%28UBX-15016656%29.pdf) state:

> To ensure GLONASS ambiguity fixing, the reference station must be configured to output RTCM message 1230. This message can be sent at a rate lower  than  the  observation  messages  as  the  bias values will  be  held  until  a  new version of the message is received. 

So this is indeed something we should enable. But I don't have the bandwidth to test that ATM.

PR: https://github.com/mavlink/qgroundcontrol/pull/5910

DonLakeFlyer, 2017-12-23:
Fixed now. @RoelHelsen86 Thanks for the report

closed 2017-12-23 by DonLakeFlyer

RoelHelsen86, 2017-12-31:
@DonLakeFlyer @bkueng 
Thanks for making and maintaining qgroundcontrol software!
Also thanks for all the feedback!