mavlink/qgroundcontrol #5892: Add support for different set of flight modes in Advanced

DonLakeFlyer, 2017-12-18:
This requires signalling support as well as change to flight mode dropdown controls to support list changing.

dagar, 2017-12-18:
Is there any interest in coordinating this with the vehicle? I'd be more comfortable actually disabling the mode rather than hiding it.

DonLakeFlyer, 2017-12-18:
Custom builds customers tend to hide lots of flight modes from the ui which is generally what they want. In non-advanced they hide most things to both disable and simplify the ui. Then in Advanced want to bring some back. For example if you have a vehicle which is not controlled with a transmitter and only missions you only want to show a minimal set of mission related flight modes as available. None of the RC control modes. So disabling from the vehicle doesn't help much.

dagar, 2017-12-19:
Makes sense, I was thinking of a few other overlapping pieces.

I'm seeing a lot more multi GCS users these days, and thinking of situations like Iridium. General users that would never use acro, offboard, etc would also want to use this mechanism. It feels more secure to actually block it and prevent potential accidents, rather than just hide it. 

I was imagining it like a bitmask of currently available modes broadcast from the vehicle. That way we could also prevent users from trying to enter position mode or hold before the vehicle is ready (valid position estimate), rather than requesting the mode and getting a generic text error back from the vehicle.

randev45, 2022-10-10:
Was support for this ever added? Working on this myself and can see some code relating to it but not sure how to use it.

closed 2026-10-04 by github-actions[bot]