mavlink/qgroundcontrol #5916: QT 5.10 maps still broken

lvale, 2017-12-27:
- Ground station operating system. - OSX 10.13.2
- QGroundControl version and build (daily, stable, self-built from source, etc.)-Master (2017-12-27)
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc. - N/A
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version. - N/A
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

Maps are unusable. Only able to see the globe image, no Zoom available

Building with QT 5.10 after seeing the PR that corrected the compilation errors on 5.10, now it compiles ok but where we can see maps the maps are frozen and unusable.

Building with QT 5.9.3 no issues.

dogmaphobic, 2017-12-28:
I have not had a chance to test this since the PR was created/merged. I will take a look at it tomorrow...

dogmaphobic, 2017-12-28:
Something is out of whack but I can readily see what is going on. Bing maps are working fine but Google maps don't even try to load. I won't have the time to look further into this in a while.

@patrickelectric Did you test all map providers with your PR?

patrickelectric, 2017-12-29:
Ok.. I can reproduce the problem..
Work with:
- [ ] Bing
- [ ] Google
- [ ] Statkart
- [ ] Eniro
[image]

patrickelectric, 2017-12-29:
The problem appears to be only with google, I'll try to solve this.

patrickelectric, 2017-12-29:
@dogmaphobic, it appears to be something related to the TileCache.
Take a look [here](https://paste.kde.org/pjusvwon8), any idea ?
It doesn't call QGeoTileFetch or anything related.

patrickelectric, 2017-12-29:
Solved, I'll send a PR asap.

PR #5923 (2018-01-05): (Fix #5916) QGeoTiledMappingManagerEngineQGC: Use cameraCaps in QGeoMapType
In Qt 5.10 QGeoMapType need QGeoCameraCapabilities as argument
E.g: https://github.com/qt/qtlocation/blob/2b230b0a10d898979e9d5193f4da2e408b397fe3/src/plugins/geoservices/osm/qgeotiledmappingmanagerengineosm.cpp#L167
Fix #5916
commit 2687df7e (2018-01-05): Merge pull request #5923 from patrickelectric/corrections

(Fix #5916) QGeoTiledMappingManagerEngineQGC: Use cameraCaps in QGeoMapType
commit c7e236c3 (2018-01-05): QGeoTiledMappingManagerEngineQGC: Use cameraCaps in QGeoMapType

In Qt 5.10 QGeoMapType need QGeoCameraCapabilities as argument
E.g: https://github.com/qt/qtlocation/blob/2b230b0a10d898979e9d5193f4da2e408b397fe3/src/plugins/geoservices/osm/qgeotiledmappingmanagerengineosm.cpp#L167
Fix #5916
closed 2018-01-05 by dogmaphobic