mavlink/qgroundcontrol #5917: structure scan does not respect distance to structure

lvale, 2017-12-28:
On certain scan of structures the distance to structure is not respected. If one works with fixed focus cameras that invalidates using the structure scan

Example mission
[image]

[acuteangle.plan.txt](https://github.com/mavlink/qgroundcontrol/files/1590320/acuteangle.plan.txt)

DonLakeFlyer, 2017-12-28:
There will be extra images, but the images that actually overlap the actual structure will be correctly distances won't they?

lvale, 2017-12-29:
It gets worst, and will have a lot of unusable pictures on each level

[image]

It's a problem when you're scanning at high resolutions,with a very narrow depth of field on the camera

DonLakeFlyer, 2017-12-29:
That is a hard problem to fix. I question whether any of these cases are reality or just strange boundary cases which nobody would ever fly?

lvale, 2017-12-29:
Very easy to fall on this, like on this struct survey of the outside walls of a small castle

[image]
[image]

DonLakeFlyer, 2017-12-30:
Tricky to fix, but I'll look in 3.4

imagidan-ets, 2024-03-08:
I suggest a parameter named Offset Resolution. Default is 0 and as you make it higher, it will round the edges more incresing the number of waypoints while minimizing distance to structure

imagidan-ets, 2024-03-08:
To do it, a resolution parameter would be added to the offset function in [src/MissionManager/QGCMapPolygon.cc](https://github.com/mavlink/qgroundcontrol/blob/master/src/MissionManager/QGCMapPolygon.cc#L430)

DonLakeFlyer, 2024-03-08:
Don't you only want to cut the corners if the angle is less than 45 degrees? Not in general.

DonLakeFlyer, 2024-03-08:
It seems like you could post-process the polygon:
* Angle less than 45 degrees
* Create new line orthogonal to the center angle of the vertex
* Position it scan distance away from the vertext
* Intersect it with the scan polygon to generate the new points to chop the edge off

closed 2026-10-24 by HTRamsey