mavlink/qgroundcontrol #5924: Empty KML files

lvale, 2017-12-29:
Using QGC master compiled with QT 5.10

Using Map provider Bing

Create a Structure scan and export to KML

Resulting file is empty, as no data inside the file

DonLakeFlyer, 2017-12-29:
@patrickelectric Can you look at this?

patrickelectric, 2017-12-30:
@DonLakeFlyer I'll try to look on it.

DonLakeFlyer, 2018-03-22:
@lvale I just tried a simple example and it worked. Do you have a .plan file that fails? I fixed a few things with KML last week so this may be good now.

patrickelectric, 2018-03-22:
@DonLakeFlyer sorry for not taking care of this, I work with underwater now, no more missions to take care

lvale, 2018-03-22:
Sorry Don
Latest build
[image]

Doesn't save either to file or to kml. When choosing save no dialog comes up

[image]

DonLakeFlyer, 2018-03-22:
What OS?

DonLakeFlyer, 2018-03-22:
Using a pre-built daily build?

DonLakeFlyer, 2018-03-22:
Oh, wait you built it yourself with Qt 5.10. We are only on Qt 5.9.3. 5.10 is likely broken.

lvale, 2018-03-23:
ok. Just re-tested master with QT5.9.4 and it is now working

DonLakeFlyer, 2018-06-20:
All working fine

closed 2018-06-20 by DonLakeFlyer