mavlink/qgroundcontrol #5925: https_proxy ignored

hsteinhaus, 2017-12-29:
QGC is partly ignoring http_proxy/https_proxy variables on Linux. The folloing behaviours are observed. QGC is unable to download any map tiles in an environment without Internet connectivity via default gateway, but with a valid proxy specified by both http_proxy and https_proxy environment variables:

```
QGeoTileRequestManager: Failed to fetch tile (140241,86642,18) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (140241,86643,18) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (140241,86641,18) 5 times, giving up. Last error message was: 'Network not available'
No Internet Access
No Internet Access
```

Surprisingly, with http_proxy and https_proxy accidentally set to a (now) invalid proxy, but *with* connectivity via default gateway, the tile download also fails with the same message. So there must be some code that evaluates the proxy setting, but that piece of code seems to be broken. 

Version of QGC: any (including latest available daily build)
Platform: Fedora 26 x86_64

DonLakeFlyer, 2017-12-29:
You know more about this network stuff than me. Maybe proxy still not quite set up right?

hsteinhaus, 2017-12-29:
Here is what I did:
```
export http_proxy="http://myproxy:3128"
export https_proxy="http://myproxy:3128"
```
After doing so, curl and wget are working, so I assume that there should be no major problem on the network side. 

I am sure QT(webkit) has some provisions for proxy support - and the negative test with the non-existing proxy in the env variable, plus a direct internet connection, seems to confirm that. Maybe some flags must be set while setting up the connection to the web server? Unfortunately I never used that part of QT by myself...

BTW, when Googling about QT and programs, you find millions of similar problems with other QT-based programs, so this must be really some standard problem.

DonLakeFlyer, 2017-12-29:
FYI: QGC code is already using Qt support for proxies.

dogmaphobic, 2017-12-29:
Proxy is an ongoing on/off problem. The main issue is that I have no way to test this as I don't have a proxy setup (nor a reason to do so).

For tile downloads, here is the code that sets up a *default* proxy. The idea being that Qt will setup the proxy based on the system configuration:

https://github.com/mavlink/qgroundcontrol/blob/master/src/QtLocationPlugin/QGeoMapReplyQGC.cpp#L159-L171

hsteinhaus, 2017-12-29:
May be the flow stops before reaching this code, e.g. here:
https://github.com/mavlink/qgroundcontrol/blob/master/src/QtLocationPlugin/QGeoMapReplyQGC.cpp#L151

dogmaphobic, 2017-12-29:
That could be the case. Here is where the code tests for Internet connectivity (attempting a connection to `github.com`):

https://github.com/mavlink/qgroundcontrol/blob/master/src/QtLocationPlugin/QGCTileCacheWorker.cpp#L1059-L1083

hsteinhaus, 2017-12-29:
Fur sure. socket.connectToHost(info.addresses().first(), 80) will clearly fail in an environment that requires that all connections go through a proxy (like many campony networks here do).

closed 2018-01-05 by dogmaphobic

DonLakeFlyer, 2018-01-05:
Could you test a new daily build?

hsteinhaus, 2018-01-05:
Unfortunately no improvement (see https://github.com/mavlink/qgroundcontrol/pull/5970)

dogmaphobic, 2018-01-05:
I don't have the bandwidth to deal with this for now. I found this within the Qt docs regarding proxies. I've never used proxies and I only have a vague understanding how it works:

http://doc.qt.io/qt-5/qnetworkproxyfactory.html

Does it means that this only works for specific requests such as HTTP or HTTPS? QGC makes other types of (direct) connections, which seems to be what is causing the issues here. If that's the case, QGC would have to identify the presence of a proxy setup and skip these (direct) tests. It's also unlikely video streaming will work if all these proxies handle are HTTP/HTTPS.

In addition, things like this don't exactly inspire a great deal of confidence:

_On Windows platforms, this function may take several seconds to execute depending on the configuration of the user's system._

hsteinhaus, 2018-01-07:
The classic proxy approach (controlled by the http_proxy/https_proxy) is mainly intended for HTTP(S) requests (with some extensions for FTP and some generalizations for generic TCP like SOCKS).

I really think that the implementation of fetching the tiles is ok, the only problem is the internet connectivity check (which is based on a direct TCP connection, instead of an abstract HTTP get. In contrast to the GET, the connect() circumnavigates the otherwise working proxy. The obvious solution would be to check the connectivity in the same way as the tiles download is done or just not check connectivity at all.

Is there really any non-HTTP protocol involved in downloading map tiles?

hsteinhaus, 2020-04-17:
Problem is still present - any chance to reopen this issue? Or should I better report a new one?

PR #5970 (2018-01-05): Attempt at using default proxy settings.
I have no way to test this. This is an attempt in dealing with proxy issues reported in #5925. Based on the documentation, this is all you need to do. In my case, with no proxy, it all worked as before.

Fix #5925