mavlink/qgroundcontrol #5932: QGC speech output buffers too much, repeats too much

px4dev, 2017-12-31:
( QGC 3.2.7 / macOS 10.13.1 / PX4 Mini firmware 1.7.2 )

In many cases, the firmware sends log messages at a rate far faster than QGC can speak them, and in many cases these messages are repeated over and over again.

This can lead to situations where QGC sits there talking to itself for half an hour after the MAV has been disconnected, or where timely information is lost behind e.g. repeated "you can't disarm in this flight mode" messages.

Messages printed by the firmware are meant to be read (because developers are looking at them in a serial terminal), not listened to; either the messages need to be rethought in terms of what should be spoken, or QGC needs to stop reading everything it receives and start interpreting messages and formatting them differently for speech.

As it stands, when things are going well, the spoken output is great, but it can't be counted on to communicate information in more challenging circumstances.

DonLakeFlyer, 2017-12-31:
> Messages printed by the firmware are meant to be read

I would disagree with that statement.

The groundstation rules for when status text is spoken are as follows:
* If the message is NOTIFY or higher severity
* If the message starts with a '#'

These rules have been the same for years now. The only thing that has changed is that the firmware isn't paying enough attention to all the stuff it spits out with respect to these rules.

LorenzMeier, 2017-12-31:
From a separation of concerns perspective the presentation of the health status is the job of the GCS. Let me make the analogy to Android: The services in Android produce hundreds of log messages in an improperly configured device (e.g. SIM card missing), but the UI filters this down to the root cause (SIM card missing).

We can't centralise the user feedback on the autopilot side without buffering all log messages, which is (again from a separation of concerns perspective) not what a safety-critical software component should be doing.

I'm currently working on a RFC for an event micro service which will take care of this problem on a fundamental level.

LorenzMeier, 2017-12-31:
Temporary fix: https://github.com/PX4/Firmware/pull/8543

closed 2017-12-31 by LorenzMeier