mavlink/qgroundcontrol #5935: Calibration timeout is too short

px4dev, 2017-12-31:
( QGC 3.2.7 / macOS 10.13.1 / PX4 Mini firmware 1.7.2 )

The sensor calibration timeout is very short, compared to the time that it takes to look at the set of remaining positions and manhandle a possibly large airframe into those positions.  

As long as the process can be cancelled, a very long (minutes) timeout would greatly reduce the amount of stress involved when wrangling cables and being furiously beeped at...

DonLakeFlyer, 2017-12-31:
Unless I'm missing something QGC isn't in control of any of the calibration timeouts. The firmware controls all of those. Can you give a specific example of the timeout? There are lots of different calibration types, each with varying timeouts within them.

LorenzMeier, 2017-12-31:
Fixed: https://github.com/PX4/Firmware/pull/8543/commits/5756c3348a47efc95595ff62c7774bb6c4d56bf8