mavlink/qgroundcontrol #5940: Airspeed cal instructions say "blow across the front", but to generate enough pressure you have to blow *into* the front of the pitot tube

px4dev, 2017-12-31:
This system is for posting bugs or feature requests ONLY. For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

----
When posting bug reports, include the following informaiton:
- Ground station operating system.
- QGroundControl version and build (daily, stable, self-built from source, etc.)
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

**All this template text should be deleted before creating your issue.**

LorenzMeier, 2018-01-01:
That sounds like there is something wrong with it. No need to blow into it.

DonLakeFlyer, 2018-01-03:
If I remember correct "across" was to prevent anything getting into the pitot which will screw i tup.

closed 2018-01-03 by DonLakeFlyer