mavlink/qgroundcontrol #5942: Too many EKF etc. errors spammed during calibration process

px4dev, 2018-01-02:
( QGC 3.2.7, Firmware 1.7.2 - this is quite possibly not a QGC problem per se and may in fact belong in the Firmware universe)

After selecting an airframe, the user will proceed to sensor calibration (QGC will almost certainly be complaining about how nothing is set up and everything is failing and the world is about to end and ...)

Before and betwee the various calibration stages, it appears as though the system attempts to initialize, and this leads to a slew of errors, which QGC dutifully attempts to vocalize. In particular, the error spam can be so bad that QGC is still complaining 10 or 20 minutes after calibration has been completed. This is particularly bad after compass and gyro cal have been completed but before accel cal is completed, as something in the firmware is looping endlessly.

LorenzMeier, 2018-01-02:
Can you please retest with 1.7.2dev (master)? That has all the fixes from your previously reported issues in.

DonLakeFlyer, 2018-01-03:
If that isn't clearing things up, you can put a new issue on the firmware side.

closed 2018-01-03 by DonLakeFlyer