mavlink/qgroundcontrol #5943: IWBNI it was clear when parameter changes are sent to the firmware / saved.

px4dev, 2018-01-02:
( QGC 3.2.7 )

It's noted in the documentation that some parameter / setting changes are saved immediately, but there is nothing in the UI, and no affordances that suggest that changes are being updated.

In addition, "things" happen when changing tabs (beeping, control surface movement, etc.) that imply that perhaps changes don't take effect until the tab is changed.

As a user unfamiliar with what QGC is doing, this leaves me wondering what I should do to ensure that my changes are saved. 

Note that I'm not necessarily suggesting the "apply / save" button approach, but some guidance in the UI (e.g. a 'saving' label that lights up when a parameter upload happens) could go a long way to guiding correct use.

DonLakeFlyer, 2018-01-03:
> It's noted in the documentation that some parameter / setting changes are saved immediately

Curios as to where that is. Almost all of QGC follows a no-save model where things are saved automatically. In the few places where an explicit save is required there will be some sort of "Save" or aptly named button to save things, and those will tend to be modal.

closed 2018-01-03 by DonLakeFlyer

hamishwillee, 2018-01-03:
> Curios as to where that is. Almost all of QGC follows a no-save model where things are saved automatically. 

I don't know where either, but I think it might be worth making an explicit statement about parameter saving here? https://docs.qgroundcontrol.com/en/SetupView/Parameters.html

Thoughts?

DonLakeFlyer, 2018-01-03:
Don't quite see how that would be too helpful. If you can't find a save button anywhere then I think most people assume that things will just work. Just like any other application which does not have save button for various UI. The parameter editor itself is onen of the very few places that actually has a save button, as does the parameter validation dialog.

hamishwillee, 2018-01-03:
It may be that you're right, but personally I wasn't certain. In any case, the parameter docs was a bit out of date, so I have updated it in https://github.com/mavlink/qgc-user-guide/pull/125 .

This includes a note: 

> **Note** When you click **Save** the parameter is automatically and silently uploaded to the connected vehicle. Depending on the parameter, you may then need to reboot the flight controller for the change to take effect.

DonLakeFlyer, 2018-01-04:
> When you click Save the parameter is automatically and silently uploaded to the connected vehicle.

They key is that happens every else as well without needing to click a save button. Like setup pages.