mavlink/qgroundcontrol #5945: MAVLink Console: UX issues

LorenzMeier, 2018-01-02:
There are a couple of UX issues with the shell:

 * Line spacing is 1.5 lines, so not a lot of content in there on a laptop screen
 * Shell scrolls instead of being locked to the bottom (the button enabling this should be enabled by default and should be sticky)
 * Text can't be copied out of it
 * As the separate text input field is unusual for a shell prepending that with `>` might help to indicate that text should go there

DonLakeFlyer, 2018-03-23:
> Shell scrolls instead of being locked to the bottom (the button enabling this should be enabled by default and should be sticky)

Don't understand what this means?

DonLakeFlyer, 2018-03-23:
Oh, wait. I had no idea "Show Latest" was essentially a checkbox, not a button.

bkueng, 2018-05-26:
Wasn't supposed to be closed yet

DonLakeFlyer, 2018-05-26:
I hit the wrong button inthe wrong place!

closed 2026-10-04 by github-actions[bot]