mavlink/qgroundcontrol #5950: Log-Download Timed Out?

Marcusqjumpfxu, 2018-01-03:
Hi there,

I use QGroundControl quite often and I've discovered an issue with the download of Log-Files.

First the download is way to much than the size of the actual Log. Second the download timed out.

I've been connected via USB to my Pixhawk. OS: Ubuntu. QGroundControl: Nightly 2017-12-25.
[image]
[image]
[image]

I've used an USB extension cable, but even without the extension cable it time out.
I use the newest stable firmware 3.5.4 ArduCopter.

The terminal gives me the following output if I try to download a log file:

```
LogDownloadLog: Request log data (id: 41 offset: 551880 size: 1080 )
LogDownloadLog: Request log data (id: 41 offset: 551880 size: 1080 )
LogDownloadLog: Request log data (id: 41 offset: 551880 size: 1080 )
LogDownloadLog: Request log data (id: 41 offset: 551880 size: 1080 )
LogDownloadLog: Request log data (id: 41 offset: 552960 size: 46080 )
LogDownloadLog: Request log data (id: 41 offset: 552960 size: 46080 )
LogDownloadLog: Request log data (id: 41 offset: 552960 size: 46080 )
LogDownloadLog: Request log data (id: 41 offset: 552960 size: 46080 )
Too many errors retreiving log data. Giving up.
```
In APM Planner 2.0.24 it works flawless.

I've posted this also on the [ardupilot discussion page](https://discuss.ardupilot.org/t/log-download-timed-out/24326).

DonLakeFlyer, 2018-01-17:
@dogmaphobic Do the size requests from the log make sense. Can't remember how that works. It suddenly bumps to a large size.

DonLakeFlyer, 2018-01-18:
I tried downloading some logs through SITL and it of course worked every time.

NaterGator, 2018-01-18:
@DonLakeFlyer if the segment the long downloader was working on had gaps (likely over any kind of less-than-perfect link) the smaller size can make sense (it would be the size of the missing contiguous gap).

DonLakeFlyer, 2018-01-18:
From my testing it looks like the 1080 are filling in gaps whereas as 46080 is the normal request size. So I'd guess there is something going wrong on the link losing data. Even over usb would be strange though.

NaterGator, 2018-01-18:
Yeah that sounds about right. I've seen the CDC-ACM interface have some loss; I'm guessing it comes from dropped transfers on the micro side but I haven't actually verified that.

DonLakeFlyer, 2018-01-19:
When you get a failed download. Do you get a status text message from the vehicle which says "Log download in progress"?

DonLakeFlyer, 2018-01-19:
@dogmaphobic Have you looked at the ArduPilot firmware which implements this? It's a bit strange. It streams out a certain number of packets them stops. While it is in the middle of streaming those packets it drops any incoming LOG_REQUEST_DATA requests from the ground station. So if the ground station asks to fill in missing gaps before while ArduPIlot is still streaming this could crap out. Not really sure how the log download code in GGC works. So not sure if this is a problem.

aklink, 2019-10-24:
Any solution? I am trying to download a 88.8 MB logfile from PX4, but always get a time out after some time. Logfiles up to 40MB in size can be downloaded flawlessly.

dimon777, 2020-02-10:
Same problem here on 71MB log. With logs bit smaller always downloaded successfully. My download speed is 165KB/s

DonLakeFlyer, 2020-02-10:
@dogmaphobic I'm not sure that this one raises to Stable but I've heard the log timeout problem multiple times along the way. Seems like this needs to be figured out if it's QGC or Firmware bug. Should the code be change to never timeout and keep retrying until the user gives up?

fredowski, 2020-02-11:
Hi, Alexey tried the log download with ardupilot over the ESP8266 wifi bridge and saw timeout problems there. The discussion is here:

https://discuss.ardupilot.org/t/pixracer-esp8266-arduplane-log-download-timeout/51850

So finally he could do the download via Missionplanner but it did not work with QGC.

DonLakeFlyer, 2020-02-12:
Hoping this is finally fixed now

closed 2020-02-12 by DonLakeFlyer

fredowski, 2020-02-13:
Hi, I reported some test results in #8312

DonLakeFlyer, 2020-02-13:
@fredowski Thanks so much for testing. I created a new Issue for this at #8325