mavlink/qgroundcontrol #5952: Switch from "Descent Rate" to "Glide Slope" in Landing Pattern editor

Kjkinney, 2018-01-03:
As mentioned in #5178 I'm moving the discussion of Fall Rate vs Glideslope to here. 

FYI @Antiheavy @M-Skelton

 >I think descent rate is the incorrect thing to use when planning a fixed wings landing path.
>
>- Descent rate suggests that the aircraft is descending at a constant rate from the last waypoint to the >land waypoint. This is false as both PX4 and Arduplane will change the descent rate to hit a glideslope.
>- Using a percentage to describe a descent rate doesn't make sense.
>- Descent rate is already used in unit/s to show how quickly the aircraft is descending at any given >time. It is also used universally in the aviation community in unit/s to describe how quickly an aircraft >will be ascending or descending
>
>
>
>I suggest that this is switched to a glideslope in degrees.
>
>- Glideslope/Glidepath is the universal aviation term when describing an airplane's path to a runway.
>- Glideslope is also used in PX4 parameters to set the fixed wings path on landing, meaning that it will >reject a mission if the glideslope is set too steep in px4. Arduplane uses glideslopes as well, however, I >do not believe they are parameter defined. This would allow them to more accurately set a glideslope >for landing rather than having to guess/do extra math.
>- Glideslope would be defined in degrees to match aviation terms, as well as to match PX4 and Arduplane parameters and documentation.

DonLakeFlyer, 2018-01-03:
@cody303 Adding Cody as well.

cody303, 2018-01-03:
That is a good idea. We work backwards to get the desired glide slope, but have to do the math.

Antiheavy, 2018-01-03:
+1 for this change

DonLakeFlyer, 2018-03-22:
Ok, in looking at this. What I discovered:
* Since you can set altitude relative independently on the landing and loiter point it makes it impossible to correctly calculate glide slope
* The current implementation of fall rate is incorrect as well due to above problem

I propose to remove the ability to independently set relative alts for both of those. And instead specify a relative altitude setting which is common to both loiter and landing point. This will allow me to calc glide slopes.

Ok with folks?

Antiheavy, 2018-03-22:
> I propose to remove the ability to independently set relative alts for both of those. And instead specify a relative altitude setting which is common to both loiter and landing point. This will allow me to calc glide slopes.

for purposes of the complex mission item (landing pattern), yes this sounds great.

Antiheavy, 2018-03-22:
fyi @dagar any thoughts?

dagar, 2018-03-22:
Sounds good to me! So will the entire landing sequence be locked together effectively?

Antiheavy, 2018-03-22:
> will the entire landing sequence be locked together effectively?

@dagar @DonLakeFlyer please note the separate discussion here https://github.com/mavlink/qgroundcontrol/issues/5674#issuecomment-358700682 about how a "lock" checkbox could work effectively along with changing the wording to "glide slope" and units to "degrees".

DonLakeFlyer, 2018-03-22:
> So will the entire landing sequence be locked together effectively?

No. Not for this specific change. It just means both altitudes (loiter, land) are relative, or both altitudes are not relative.

Regarding #5674, that is a good change as well. I'll lump that together with this and work it all out together.

DonLakeFlyer, 2018-03-23:
Another assumption: Changing Glide Slope leaves the landing point the same and move the loiter point in or out to achieve the requested glide slope.

Antiheavy, 2018-03-23:
Yes. I also think if a “lock dist” checkbox is implemented then changing the glide slope should auto-unlock the distance. As appears to changing the loiter altitude. This seems safer when the assumption is the default loiter altitude is typically above trees/power lines.

Antiheavy, 2018-03-23:
“As appears” should be “as opposed”

DonLakeFlyer, 2018-03-23:
The Glide Slope change ended up getting a bit involved so I didn't put "Lock" in there as well. That will come in separately.

DonLakeFlyer, 2018-03-23:
Daily builds should kick out in a hour or so. Could use some extra testing if folks have the time.

cody303, 2018-03-23:
Sorry for the newbie question, but where can we find the daily builds?

Antiheavy, 2018-03-23:
> Sorry for the newbie question, but where can we find the daily builds?

this is where I download then from: https://docs.qgroundcontrol.com/en/releases/daily_builds.html   beware, don't use the daily builds for flying or really anything other than ground testing IMHO...

Antiheavy, 2018-03-25:
I'm finding some issues with this in the 2018-03-23 Daily build.

[image]

If i plan the landing pattern it looks like this:
[image]

If I upload I get then "...move away by infinity." warning.

then if I "download" the mission off the autopilot it shows up with the Land waypoint at the same height as the other waypoints.
[image]

Antiheavy, 2018-03-25:
Hmmm....  actually, now I'm not getting the infinity error.  I reset all parameters on my autopilot to defaults, re-flashed firmware, then manually set my parameters back to my desired values.  I reverted to an older QGC, then re-installed the latest Daily build.

The Land waypoint altitude graphic still shows incorrectly when the mission is downloaded off the autopilot though:
[image]

DonLakeFlyer, 2018-03-25:
Complex Mission items show the height of the entry point in the comlpex item in the mission item status display at the bottom. So the "L" shows the heigh of the Loiter altitude.

DonLakeFlyer, 2018-03-25:
For the case where the firware rejects I'm going to need the specific plan file.

DonLakeFlyer, 2018-03-25:
Also can you move to a separate issue?

Antiheavy, 2018-03-29:
> For the case where the firware rejects I'm going to need the specific plan file.
after reseting all firmware params to defaults, reflashing firmware fresh, then reapplying parameters, I can no longer reproduce this error.  So it must have been some weird firmware state I got myself into...

Antiheavy, 2018-03-29:
and more to the point of this issue:  the "Glide Slope" selector seems to be working well in the recent Daily Builds:
[image]

Antiheavy, 2018-04-04:
@DonLakeFlyer is this issue still open, or are you happy with how it is in the daily builds now?

DonLakeFlyer, 2018-04-04:
Should be closed

closed 2018-04-04 by DonLakeFlyer