mavlink/qgroundcontrol #5978: Import KML/KMZ and SHP to missions

lvale, 2018-01-06:
There's already support to export KML files from missions. Would be great to have support to also import KML/KMZ and additionally SHP files either for missions and/or fences

@DonLakeFlyer

ThorstenDP, 2018-01-11:
Importing SHP files would be awesome!
There is open source ( "MIT Style" license, or at the option of the licensee under the LGPL) library for reading and writing shapefiles available at: http://shapelib.maptools.org/

Antiheavy, 2018-01-17:
+1 for shape file support (SHP).  also see: https://github.com/mavlink/qgroundcontrol/issues/5752

closed 2025-05-14 by HTRamsey

mrschick, 2025-05-15:
I have developed a rough implementation of a KML import for a current project of mine.
ATM it expects KML files to provide altitude information for the waypoints, so imported files must follow the `lon,lat,alt` format, or it will crash on import.
https://github.com/MAVTech-srl/qgroundcontrol/compare/mavlink%3Aqgroundcontrol%3Amaster...feature/kml-wp-import

I guess the cleanest solution would be letting the "Open" button import a plan from any acceptable format, determining the parsing function internally.

mrschick, 2025-05-15:
I believe I looked over it and concluded that its core use was exporting a plan to KML, and so doing a straight port of the .plan import function to KML was more appropriate.