mavlink/qgroundcontrol #5981: QGC doesn't want to flash while FTDI cable is connected

AndreasAntener, 2018-01-06:
The firmware tab tells me I should disconnect the Pixhawk/Radio. I understand why the FTDI cable gets detected, but I don't think the it needs to listen for modem connections in the firmware tab? Could that be separated?

DonLakeFlyer, 2018-01-07:
Don't quite understand. You have a SiK Radio connected or something else?

AndreasAntener, 2018-01-07:
I have an FTDI cable connected, always ;). Mostly for debugging or something else. The Pixhawk is of course connected via USB. In order to flash I now need to disconnect the FTDI cable (and then reconnect it again because I need it after all..). This is a bit of a corner case, but maybe the port detection while flashing could ignore things like FTDI cables.

DonLakeFlyer, 2018-01-07:
> This is a bit of a corner case, but maybe the port detection while flashing could ignore things like FTDI cables.

Not quite sure since some SiK Radios show up as FTDI.

AndreasAntener, 2018-01-07:
But AFAIK you don't flash firmware via those?

DonLakeFlyer, 2018-01-07:
Not quite sure. @LorenzMeier Do you know?

AndreasAntener, 2018-01-07:
I know it's possible, there is still a PR from me on the bootloader to do this ;). But it's really not practical as it would take 5 to 10 minutes with the standard gear..

Or, does QGC support flashing the SiK radios?

DonLakeFlyer, 2018-01-07:
QGC supports flashing SiK radios.