mavlink/qgroundcontrol #5995: Force user to adjust hover throttle

LorenzMeier, 2018-01-08:
@DonLakeFlyer I don't know yet what the best way is to do this, but it would help if the user was guided to the tuning dialog during setup. Would it be possible to color tuning as red based on the first param MPC_THR_HOVER == default? It should be adjusted per airframe.

https://github.com/PX4/Firmware/issues/8625

Any other recommendations how we could raise with users that they need to adjust this? Should we include more / better instructions?

DonLakeFlyer, 2018-01-15:
Is MPC_THR_HOVER adjusted by SYS_AUTOSTART script values based on airframe? If so, then I don't really know what the default is.

DonLakeFlyer, 2018-01-16:
I don't think I'll be able to get this into 3.3. I requires more than just triggering component set now complete. It required the tuning page itself to show which tuning sliders needs changing. Not hard, I'm locking down hard on 3.3 so I don't spend the rest of my life trying to get it out.