mavlink/qgroundcontrol #5996: Ubuntu 16.04, Segfault on Startup in Release Build

acfloria, 2018-01-10:
When building the latest master in the release mode I get a SEGFAULT on the startup. However, the debug build runs without an issue.

I tested some different commits to identify when it broke. It still works with commit `ce0e3ad06b6b7930851102281c16bfc78f97045d`. From commit `40708809d02cfae77a1f4244a2ab40f93769ea22` on I get the issue. So I suspect that somewhere in between those commits the error is introduced.

From GDB I get the following output:
```
#0  QtPrivate::RefCount::ref() (this=0x0) at ../../../Qt/5.9.2/gcc_64/include/QtCore/qrefcount.h:55
#1  0x000000000055dab8 in QString::QString(QString const&) () at ../../../Qt/5.9.2/gcc_64/include/QtCore/qstring.h:907
#2  FactMetaData::FactMetaData(FactMetaData::ValueType_t, QObject*) (this=this@entry=0x1014720, type=valueTypeInt32, parent=parent@entry=0xff95a0 <ParameterManager::_defaultFact>)
    at ../qgroundcontrol/src/FactSystem/FactMetaData.cc:120
#3  0x000000000066b170 in Fact::Fact(QObject*) [clone .constprop.2543] (this=this@entry=0xff95a0 <ParameterManager::_defaultFact>, parent=0x0) at ../qgroundcontrol/src/FactSystem/Fact.cc:29
#4  0x00000000004483e8 in __static_initialization_and_destruction_0 (__initialize_p=1, __priority=65535) at ../qgroundcontrol/src/FactSystem/ParameterManager.cc:37
#5  _GLOBAL__sub_I_ParameterManager.cc.lto_priv.3430 () at ../qgroundcontrol/src/FactSystem/ParameterManager.cc:1482
#6  0x0000000000448827 in _GLOBAL__I_65535_0_QGCMapEngine.o ()
#7  0x00000000006987cd in __libc_csu_init ()
#8  0x00007ffff7a2d7bf in __libc_start_main (main=0x446860 <main>, argc=1, argv=0x7fffffffdb98, init=0x698780 <__libc_csu_init>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdb88) at ../csu/libc-start.c:247
#9  0x0000000000448fe9 in _start ()
```

AndKe, 2018-01-11:
not sure how related it is, but I too am unable to run master.
Ubuntu 17.10 Qt5.9.2   (few weeks ago, I verified that 5.9.3 worked perfectly as well..)
```
Starting program: /home/andre/prog/qgroundcontrol/build/release/QGroundControl 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
QtPrivate::RefCount::ref() (this=0x0)
    at ../../../Qt/5.9.2/gcc_64/include/QtCore/qrefcount.h:55
55	        int count = atomic.load();
(gdb) q

```

acfloria, 2018-01-11:
I see, then I will try Qt5.9.3 too tomorrow and report if it then works for me. In the installation guide it is specified that one should use 5.9.2 though.

AndKe, 2018-01-11:
I doubt it will. I used 5.9.2 just fine, then tried 5.9.3 , (just fine)   ..  since christmas it stopped working, and reverting to 5.9.2  - or trying another computer does not help. (I verified this on two computers.)

AndKe, 2018-01-12:
Additional info:
I started a laptop that has built QGC from master in late December.
- it worked fine, copied the release directory.
- updated Ubuntu 17.10 , rebooted
-pulled upstream master, rebuilt.
the newly build QGC crashes.
the backed release directory with master from December runs fine.

So it's not likely to be caused by the OS or libraries ...

dogmaphobic, 2018-01-12:
I think I found it. I run a Windows build today and had the same crash with the FactMetadata constructor. It looks like it doesn't like the use of the `tr()` macro on a static object used in the constructor.

PR #6001 (2018-01-12): Fixed crash in FactMetadata constructor.
It did not like the use of the tr() macro within a static QString in the constructor.

The fix may be overkill but I didn't have the time to try different things (it takes forever to get a Windows build running for me)

Fixes #5996
closed 2018-01-12 by DonLakeFlyer