mavlink/qgroundcontrol #5997: Daily build (11.01.2018) crashes on Samsung Galaxy Tab A 7.0 (T280N)

ThorstenDP, 2018-01-11:
Ground station operating system.

- Android

QGroundControl version and build (daily, stable, self-built from source, etc.)

- Daily: 11.08.2018

Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

- Crashes on start with or without telemetry connected
- 3.2.7 works properly

dagar, 2018-01-12:
Reports from other users 

> my QGC(Daily build) crashed with Jan 7th even Jan 8th update,could send me a reverse build?like Jan 6th or before,please?

dogmaphobic, 2018-01-12:
I found (and fixed) a start up crash related to FactMetada construction (#6001)

DonLakeFlyer, 2018-01-12:
Might be. I doesn't crash for me on a 5.9.3 debug build I made. But daily does crash for me.

dogmaphobic, 2018-01-12:
It doesn't crash at all on macOS. I only saw it when building for Windows.

DonLakeFlyer, 2018-01-12:
I merged in your FactMeta data change and then pulled the built apk from S3 and it worked. So looks like that is it. @ThorstenDP I think I may take a little more time to work through play store. Can you try to install a daily from tomorrow and verify?

ThorstenDP, 2018-01-12:
@DonLakeFlyer it is already there! With fee0d4d12 it is working again. I've tested it on two devices.
Thanks for all your hard work!

closed 2018-01-12 by ThorstenDP