mavlink/qgroundcontrol #6003: Cannot update firmware

joe-marrone, 2018-01-12:
Don:
I posted this previously and have not been able to solve it, and nobody on the PX4 Discuss forum has any solution.
It boils down to the fact that QGC will not show me a screen that prompts me to update the firmware. 
I hope the post below is clear.
Can you help?

Joseph_Marrone

When the Pixhawk is connected to QGC using USB, I can see the entire setup menu:
Summary
Firmware - this says to unplug the PH from USB
Radio
Sensors
Flight modes
...
...
...
All my sensors are calibrated (green). QGC shows the position and attitude of the Pixhawk. Everything looks good at this point.

When I disconnect the USB cable I loose connection (this is normal), and the setup menu now shows only Summary and Firmware tabs (this too, is normal). There is nothing in the Summary tab except the statement "vehicle settings and info will display after connecting your vehicle. Click Firmware on the left to upgrade your vehicle."

When I click Firmware tab I get the message:
"QGroundControl can upgrade the firmware on Pixhawk devices, SiK Radios and PX4 Flow Smart Cameras.
All QGroundControl connections to vehicles must be disconnected prior to firmware upgrade.
Please unplug your Pixhawk and/or Radio from USB."

At this point I already have the Pixhawk unplugged!!! It does not prompt me to plug it in to do the firmware update. When I plug it in, I'm now back where I started from. I never see the menu for selecting a device to update.

Thanks,
Joe

DonLakeFlyer, 2018-01-12:
Using the instructions here: https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html. Turn on FirmwareUpgradeLog. Then post the log for the case where you go to Firmware tab without anything plugged in.

joe-marrone, 2018-01-12:
What's the best way to copy the log screen.  I can't highlight it for copying.

joe-marrone, 2018-01-12:
[image]
 I tried the snipping tool and pasted a copy here.  I hope this works.

DonLakeFlyer, 2018-01-12:
First step, use a daily build instead. See if it still happens. Daily builds also have a button to save log files to a file.

joe-marrone, 2018-01-12:
I thought I installed the daily build already.  I'll try again.

patilakash, 2019-06-26:
I also got same Issue in QGCS. Firmware Loading is working in Mission Planner but not in QGCS. The log file is attached. Please suggest me the solution
[Firmware_Issue_Log.txt](https://github.com/mavlink/qgroundcontrol/files/3328421/Firmware_Issue_Log.txt)

DonLakeFlyer, 2019-06-26:
> I also got same Issue in QGCS.

What exact issue?