mavlink/qgroundcontrol #6016: vehicle current command display

dagar, 2018-01-14:
**Feature request**
Display vehicle's current position setpoint graphically outside of a MISSION. This includes showing a loiter with radius and direction in HOLD mode, or parts of RTL, or often just a regular waypoint. 

In an autonomous mission QGC already has the items locally and the user can see what the vehicle is trying to do. In other autonomous modes like HOLD (in place or MAV_CMD_DO_REPOSITION), Return/RTL, the vehicle might be headed to a waypoint or loiter, but the user has no visual indication.

Can we improve this? Is it possible with the current mavlink spec, or do we need new messages?

This is potentially a broader discussing that needs to happen at the mavlink level, but I thought I'd start on the side it's actually needed. 

@DonLakeFlyer @antiheavy @LorenzMeier

Antiheavy, 2018-01-14:
other potentially related issues posted in this repository:
https://github.com/mavlink/qgroundcontrol/issues/5748
https://github.com/mavlink/qgroundcontrol/issues/4204
https://github.com/mavlink/qgroundcontrol/issues/3605
https://github.com/mavlink/qgroundcontrol/issues/2744

... I remember seeing others too, but github search isn't turning them up.

Antiheavy, 2018-01-17:
Another thing I've found very useful in the past with other systems has been commanded altitude vs current altitude.  I've seen this handled as simply as just two text values displayed near each other and labeled (e.g. at it's simplest this could be just another telemetry widget value).  I've also seen heads-up-display or artificial horizon multi function displays that incorporate visual current vs commanded altitude.

Antiheavy, 2018-10-09:
related: https://github.com/mavlink/qgroundcontrol/issues/6721

closed 2026-10-05 by github-actions[bot]