mavlink/qgroundcontrol #6018: Fly/Action button greyed out - always

zs6buj, 2018-01-15:
Issue moved from Dronecode/Using QGroundcontrol as requested

Latest Daily Build on Samsung Tab S2 9.7 on Android 7.0
Pixhawk 1 running PX4 1.7.2
Telemetry - ESP8266 compiled and flashed with https://github.com/dogmaphobic/mavesp8266
Distance: Within about 20m/30m.

Issue: After connecting successfully from QGC, all parameters appear to load and scrren information good and responsive. I arm and takeoff, then expect to fly real-time, but since the Fly/Action button remains greyed out I can't change altitude or fly by touching points on the screen. Fortunately I can RTL.

More information here: http://discuss.px4.io/t/fly-action-button-greyed-out-always/5345

DonLakeFlyer, 2018-01-17:
Can you do this:
* Uninstall QGC
* Install this apk: https://s3-us-west-2.amazonaws.com/qgroundcontrol/DonLakeFlyer/android-build-debug.apk
* Get to the point where actions should be enabled
* Save Console log output to a file
* Zip it up and attach here

zs6buj, 2018-01-18:
Ok, this debug version of QGC is insisting I re-calibrate everything, including flight modes, while the latest Windows version is happy. I'll do this a little later today.

[image]

zs6buj, 2018-01-18:
Updated Pixhawk to 173, then tried to re-calibrate everything through the debug version on QGC. Started with airframe, as it is simple, but the PX4/QGC does not remember the setting after reboot.

Went back into the latest Widows version of QGC (version shown above) and re-calibrated everything.

This is what I get:
[image]

zs6buj, 2018-01-18:
Opps, Closed in error.

DonLakeFlyer, 2018-01-18:
Something is very out of whack. Looks like your parameters are not loading correctly. Did you get an error about parameter load?

zs6buj, 2018-01-19:
I re-calibrated using the daily build and the debug version starts ok now. Strange. Anyway I then did as you requested:
Connected to craft.
Pressed safety switch.
Clicked arm and pushed slider over to the right.
Craft armed.
Clicked Take-off and pushed slider over to the right.
Craft takes off, hovers.
Fly/Action button remains greyed out.
Land etc using RC radio.

The Console log as attached.

Thanks again for your help.

The Console log is here:  
[AppLog01.zip](https://github.com/mavlink/qgroundcontrol/files/1645689/AppLog01.zip)

DonLakeFlyer, 2018-01-19:
@zs6buj The problem is that QGC isn't getting the state information across telemetry that the vehicle is flying. It thinks it is still on the ground. Does this only happen with an ESP8266 connection? Do you know if the same vehicle vehicle with an SiK Radio connection? Do you know if those only repros overe WiFi on android? What about a desktop build?

@dagar Do you know if the EXTENDED_SYS_STATE mavlink message is streamed? Or is only sent from the vehicle on state change? If it is only sent on state change then it could easily be lost over a lossy connection like WiFi.

zs6buj, 2018-01-20:
I flew the same craft using udp wifi and android some time ago. Maybe 18 months. Is EXT_SYS_STATE a fairly new development?

I'll do some more checking. First I'll test with 433MHz SiK radios, and possibly with an OTG and android, otherwise the Windows version.

Second, I want to check if mavesp8266 (On GIT link mentioned above) is data (content) agnostic. i.e. will it pass the #245 message

zs6buj, 2018-01-20:
Ok, I just tested with my old 3DR radios, and it works fine. Sorry for the red herring and thanks again for listening.

I'll go away and check why the WiFi link does not pass the #245.

DonLakeFlyer, 2018-01-20:
One thing is I'm not sure the wifi firmware supports mavlink 2. 

You can check whether QGC is getting EXTENDED_SYS_STATE  message if you use a desktop build and the mavlink inspector widget plus your wifi setup. You should see EXTENDED_SYS_STATE.landed change from 1 to 2. This indicates the vehicle is flying and will enabled the other commands. I just tested this myself using SITL and firmware master. The EXTENDED_SYS_STATE message should be streamed to QGC once a second. So I can't see how it could be missed.

zs6buj, 2018-01-21:
The mavlink inspector is a huge help. I can see immediately that the EXTENDED_SYS_STATE message is never making it to QGC when I use dogmaphobic/mavesp8266, but does get through consistently using the SiK radios.  In fact, the ESP8266 implementation drops a few message types, including HOME_POSITION #242 and EXTENDED_SYS_STATE #245. I can't believe that packet loss would account for selectively dropping message types 100% of the time.

SiK Radios:
[image]

dogmaphobic/ESP8266:
[image]

DonLakeFlyer, 2018-01-21:
@dogmaphobic Any ideas as to why?

dogmaphobic, 2018-01-21:
The mavesp8266 doesn't "filter" anything. It doesn't quite make sense to have some messages come fine and others not. With that said, the MAVLink submodule in that repo hasn't been updated in ages, though these messages (242 and 245) have not been changed.

Needless to say, mavesp8266 doesn't handle MAVLink V2.

DonLakeFlyer, 2018-01-21:
@zs6buj I think you are going to need to talk with the firmware folks. Only thing I can think is that the firmware for some reason isn't streaming EXTENDED_SYS_STATE when it is talking over a udp connection. No idea why. But there isn't anything QGC can do to fix your problem. It is all working correctly according to the information it is getting.

closed 2018-01-21 by DonLakeFlyer

dogmaphobic, 2018-01-21:
> for some reason isn't streaming EXTENDED_SYS_STATE when it is talking over a udp

Both the mavesp8266 and the SiK radio use a (same) serial connection to the Pixhawk. The difference is the bandwidth and data rate. I'm not sure what "special" things the autopilot does for each link.

DonLakeFlyer, 2018-01-21:
The firmware side does some amount of rate limiting I believe on these values which are streamed based on bandwith. So if you have a crappy connection it may rate limit some things out of existence not sure. Only the firmware folks would know for sure.

zs6buj, 2018-01-22:
Ok, I realise it's time to take this issue elsewhere.

But interestingly, when I switch to a cheap commercial wifi bridge [like this](https://www.aliexpress.com/item/Newest-Wireless-WIFI-Radio-Telemetry-for-APM-2-6-Pixhawk-PX4-Replace-Traditional-3DR-Telemetry-Support/32408772542.html?ws_ab_test=searchweb0_0,searchweb201602_2_10152_10151_10065_10344_10068_10342_10343_10313_10059_10340_10341_10534_100031_10084_10604_10617_10083_10616_10103_10307_10615_10142,searchweb201603_25,ppcSwitch_5&algo_expid=3a0fe2c2-2fcb-414e-aa72-9eacaaae4f70-0&algo_pvid=3a0fe2c2-2fcb-414e-aa72-9eacaaae4f70&transAbTest=ae803_4&priceBeautifyAB=3) then all the messages seem to get through. One noteworthy difference is that this board uses TCP rather than UDP, so there is some error correction. But I'm still not convinced that the error devil targets only certain message types with 100% consistency.
APM_PIX_WIFI
[image]

zs6buj, 2018-01-22:
Fixed! I updated the mavlink libraries for mavesp8266 to a later version,   
[mavlink.zip](https://github.com/mavlink/qgroundcontrol/files/1651722/mavlink.zip)
cleaned and recompiled mavesp8266. These libs might not be completely up-to-date.
[image]

Thanks again for helping me find the solution.

Now we can definitely close the issue :)

dogmaphobic, 2018-01-22:
> I updated the mavlink libraries for mavesp8266 to a later version

Yes, that had not been updated in over a year. It does bring another question though. Isn't this supposed to be "forever" compatible? How can a later change affect an existing implementation?

DonLakeFlyer, 2018-01-22:
Not a good thing. Shouldn't need to update that firmware.