mavlink/qgroundcontrol #6020: Vagrant can't find box 'boxcutter/ubuntu1604'

pieter-pozyx, 2018-01-15:
- Operating System: Lubuntu 17.04 Zesty
- QGroundControl version: Latest version ([commit 8e16b264cd208c74e3bb03ce3dd462dee9307e33](https://github.com/mavlink/qgroundcontrol/tree/8e16b264cd208c74e3bb03ce3dd462dee9307e33))

**Steps to reproduce the issue:**
Follow the [readme](https://github.com/mavlink/qgroundcontrol/tree/8e16b264cd208c74e3bb03ce3dd462dee9307e33):
- Download and install Vagrant
- Clone from git: `git clone --recursive https://github.com/mavlink/qgroundcontrol.git`
- run `vagrant up`

Vagrant can't find the box "boxcutter/ubuntu1604":

```
vagrant up
Bringing machine 'default' up with 'virtualbox' provider...
==> default: Box 'boxcutter/ubuntu1604' could not be found. Attempting to find and install...
    default: Box Provider: virtualbox
    default: Box Version: >= 0
The box 'boxcutter/ubuntu1604' could not be found or
could not be accessed in the remote catalog. If this is a private
box on HashiCorp's Vagrant Cloud, please verify you're logged in via
`vagrant login`. Also, please double-check the name. The expanded
URL and error message are shown below:

URL: ["https://vagrantcloud.com/boxcutter/ubuntu1604"]
Error: The requested URL returned error: 404 Not Found
```

pieter-pozyx, 2018-01-15:
I was able to build using Vagrant by changing `config.vm.box = "boxcutter/ubuntu1604"` into `config.vm.box = "ubuntu/xenial64"` in the "Vagrantfile"

DonLakeFlyer, 2018-01-15:
@dagar Do you know anything about vagrant?

closed 2018-01-16 by DonLakeFlyer

dagar, 2018-01-17:
No I don't know anything vagrant. If we want to support it I think it needs to be a part of CI.