mavlink/qgroundcontrol #6028: Qgroundcontrol freezes under win8.1

hicham1492, 2018-01-15:
Hi;

The Qgroundcontrol keeps freezing (not responding) after a few munites of use.
Below are the details of the OS version, Qgroundcontrol version, autopilot board and autopilot firmware, then the steps to reproduce the issue.

- Ground station operating system:     Windows Embeded 8.1 Industry Pro running on ZEBRA ET55 tablet
- QGroundControl version and build:  version 3.2.7 latest downloaded on 09Jan2018
- Autopilot board:                                 Pixhawk 1
- Autopilot firmware:                            Installed 'ArduPilot Flight stuck' ArduSub V3.5.2

- Exact steps to reproduce the problem. 
   1- Start Qgroundcontrol
   2- Plug in power to the ROV
   3- All connected correctly, we then receive video stream from the ROV onto Qgroundcontrol. 
   4- We run some few tests, like running the thrusters, moving the camera and confirming the video stream is correct, moving the map, all happy.
   5- After a while, could be few munites, the qgroundcontrol freezes. The video stream with the map freeze and message says "application not responding.

When watching the qgroundcontrol process on the task manager, the memory allocated to qgroundcontrol keeps growing, when it reaches 1.4MB a popup comes up saying "(QGroundControl.exe:3828): GLib-ERROR **:gmem.c103: failed to allocate 96091 bytes"

I have taken a process file dump, I am not sure where to upload it as it is around 970MB.
I have switched on and saved console log for all the modules. The last logging at the time of the freeze is shown below.

Any help in finding the reason behind this issue will be appreciated.
Is the issue is due to the operating system I am running on?
Or, the tablet I am using?
Or something else?

Many thanks
Hicham

[D] at ..\src\comm\LinkManager.cc:487 - "-----------------------------------------------------"
[D] at ..\src\comm\LinkManager.cc:488 - "portName:           "COM1""
[D] at ..\src\comm\LinkManager.cc:489 - "systemLocation:     "\\\\.\\COM1""
[D] at ..\src\comm\LinkManager.cc:490 - "description:        "Communications Port""
[D] at ..\src\comm\LinkManager.cc:491 - "manufacturer:       "(Standard port types)""
[D] at ..\src\comm\LinkManager.cc:492 - "serialNumber:       """
[D] at ..\src\comm\LinkManager.cc:493 - "vendorIdentifier:   0"
[D] at ..\src\comm\LinkManager.cc:494 - "productIdentifier:  0"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\comm\LinkManager.cc:487 - "-----------------------------------------------------"
[D] at ..\src\comm\LinkManager.cc:488 - "portName:           "COM1""
[D] at ..\src\comm\LinkManager.cc:489 - "systemLocation:     "\\\\.\\COM1""
[D] at ..\src\comm\LinkManager.cc:490 - "description:        "Communications Port""
[D] at ..\src\comm\LinkManager.cc:491 - "manufacturer:       "(Standard port types)""
[D] at ..\src\comm\LinkManager.cc:492 - "serialNumber:       """
[D] at ..\src\comm\LinkManager.cc:493 - "vendorIdentifier:   0"
[D] at ..\src\comm\LinkManager.cc:494 - "productIdentifier:  0"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"
[D] at ..\src\Joystick\Joystick.cc:521 - "name:roll:pitch:yaw:throttle "XInput Controller #1" 0 0 0.0390935 0.5"

jaxxzer, 2018-01-16:
Can you put the dump file on dropbox or google drive and put a sharing link here?

hicham1492, 2018-01-16:
Thanks;
I have upload it in google drive, see link below.

https://drive.google.com/file/d/1KkhLB28syD70GRyOEf12PzF15bjxLGfY/view?usp=sharing

Regards
Hicham

hicham1492, 2018-01-16:
These are the links again with the log file too and some snapshots of the task manager.

https://drive.google.com/open?id=1XGnrvXDx6uz4qBLVSXmi6KGF3jnGv7uR
https://drive.google.com/open?id=1RSWTCKWdTmlOO1JB5rmh92ucSZ0oesqt
https://drive.google.com/open?id=1KkhLB28syD70GRyOEf12PzF15bjxLGfY
https://drive.google.com/open?id=13NcNuEJM6EZTazBII94IiGTVmIu2jtrR

hicham1492, 2018-01-26:
Just would like to add more details of the tablet we are having the issue with:

> Intel Atom CPU Z3795 1.6Ghz
Display adapter: Intel HD Graphics
Display driver : igdumdim64, igd10iumd64, ...,igdumdim32, igd10iumd32.
Memory 4GB
Win 8.1 Embeded Industry Pro
64bit x64bit processor
> 

By all means I am ready to test any code fix that you could provide on the device I am using.

an question on the side: "What would be the best computer/tablet specification that the QGC works perferct?"

jaxxzer, 2018-02-05:
Not much useful in the call stack. I've no idea what to make of this.
[image]

@DonLakeFlyer is it possible to switch to Qt 5.9.4 or 5.10.0 for Stable 3.3? Then we will have debug symbols for the Qt libraries.

https://bugreports.qt.io/browse/QTBUG-3934
https://www.appveyor.com/docs/build-environment/#qt

DonLakeFlyer, 2018-02-05:
3.3 is pretty much done. Switching to a new Qt is a royal PITA.

DonLakeFlyer, 2018-02-05:
> GLib-ERROR **:gmem.c103: failed to allocate 96091 bytes"

This would seem to indicator gstreamer doesn't it. Not sure any other code in QGC uses GLib? Anyway is the memory leaking not reproducible? I would tackle that first. Try turning off video and see if the leak goes away.

hicham1492, 2018-02-13:
I have run few tests with the video stream off, and the issue still happened.
The memory leak in this case is hapening slowly (by me watching the task manager) compared to when the video stream was on.
I have uploaded a new dump file while the video stream was off

https://drive.google.com/open?id=1xg6qh_widWJ7IrJDSAjqcK4oYfovAW9e

hicham1492, 2018-02-13:
Actually, the memory did increase to 348MB but QGC didn't crash and no message showed up with the Glib-ERROR as before.

hicham1492, 2018-02-20:
Just to clarify, when I said QGC didn't crash, I meant it stopped responding but did not crah with the memory leak.

lida2003, 2024-05-09:
QGC is very slow to launch, and switch to setting tabs.  Any one knows why?

```
Edition	Windows 11 Pro
Version	22H2
Installed on	‎23/‎5/‎31
OS build	22621.3527
Experience	Windows Feature Experience Pack 1000.22700.1003.0

Device name	Daniel-SurfacePro
Processor	Intel(R) Core(TM) i5-7300U CPU @ 2.60GHz   2.71 GHz
Installed RAM	8.00 GB
Device ID	C5FC645F-69FA-4372-A8F8-CB8C596953A3
Product ID	00330-80000-00000-AA147
System type	64-bit operating system, x64-based processor
Pen and touch	Pen and touch support with 10 touch points

```