mavlink/qgroundcontrol #6039: Display and Edit Polygon PX4 Geofence in QGC

deonblaauw, 2018-01-17:
I followed the instructions described [here](https://pixhawk.org/users/geofence) to setup a polygon geofence for PX4. I've been testing it using Gazebo in HITL and it works great. When I try to add waypoints outside of the geofence, PX4 does not allow it. During a flight, when the vehicle violates the geofence, it also works really well.

The only thing I'm missing, is actually seeing the geofence visually in QGC. It would be great if we could not only view, but also edit the polygon PX4 geofence using QGC.

I'm using QGC v3.2.7 and PX4 v1.7.0.

DonLakeFlyer, 2018-01-17:
Already supported. Use a daily build instead of a stable build. Doc: https://docs.qgroundcontrol.com/en/releases/daily_build_new_features.html

closed 2018-01-17 by DonLakeFlyer

deonblaauw, 2018-01-17:
Nice, thanks! Will give it a try!