mavlink/qgroundcontrol #6047: Help with QGroundControl shortcuts

hicham1492, 2018-01-17:
Hi;

QGroundControl installs 3 shortcuts, QGroundControl, QGroundControl (GPU Safe Mode), and QGroundControl (GPU Compatibility Mode). What each one is used for?

thanks

DonLakeFlyer, 2018-01-17:
They are used when you are having problems with OpenGL. QGC uses Qt which in turn uses OpenGL graphics drivers for display the user interface. On windows OpenGL drivers have a history of flakiness. Especially on laptops with Intel chipsets.

GPU Compatability Mode - This cause QGC to yse ANGLE drivers which implement OpenGL on top of DirectX. This should be the first thing to try if you have ui visual issues or crashes in video drivers.

GPU Safe Mode - The causes QGC to not use any OpenGL driver but instead to use a software rasterizer. This should be the last resort since it will run much more slowly.

@hamishwillee Would be good to get this in the docs somewhere.

closed 2018-01-17 by DonLakeFlyer

hicham1492, 2018-01-18:
Many thanks for the clarification.

In one of the Desktop running Win10 we couldn't run the first shortcut. We never got it to run as as soon as we click on the first shortcut a popup comes up saying this application can't run.
So, we start using the second shortcut and it worked just fine with no issues.
So in short, my desktop doesn't run OpenGL, it needs ANGLE driver to implement it on top of whatever driver my desktop is running.

As you mentioned, would be good to put this as part of the system requirement. i.e. the optimum system requirements "Graphic driver running OpenGL"