mavlink/qgroundcontrol #6052: FTDI autodetect of RTK base station

superjax, 2018-01-19:
I have an InertialSense RTK base station, which uses the same FTDI chip as the 3DR radios to communicate over USB.  I've extended the GPS driver to support the inertialsense driver, but I have to hack the `json/USBBoardInfo.json` file, so that it can be detected as an RTK GPS, and not as a radio.

This is obviously a problem if I want to use both a radio and the inertialsense, (which is the only way that the RTCM inject feature of QGroundControl is useful).  I'm happy to make the modifications to QGroundControl myself, but I would love some guidance on how to resolve the ambiguity between the two devices, given that they have the same product ID and vendor ID.

Does anyone have any ideas on how this might be accomplished?

DonLakeFlyer, 2018-01-19:
You are going to have to turn off Auto-connect for SiK radios and specify you radio connection by port manually. Then the RTK can autoconnect, and you manually connect the SiK Radio.

superjax, 2018-01-19:
Sounds good!  Thanks!

closed 2018-01-19 by superjax