mavlink/qgroundcontrol #6053: QGC wont start (windows 10)

profff, 2018-01-20:
hello here's the issue: 

- Ground station operating system.
windows 10 on acer aspire switch 
- QGroundControl version and build (daily, stable, self-built from source, etc.)
just last stable release from here https://s3-us-west-2.amazonaws.com/qgroundcontrol/latest/QGroundControl-installer.exe
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
dont relevant I have not plugged it yet (piwhawk 2.1)
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
dont relevant 
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.
just stock installation and try to start from quick links in start menu
i tried standard link and both gpu safe and compatibility mode nothing starts 
no error messages just few seconds of timer mouse icon and back to nothing 
I tried also to repair or uninstall and reinstall it nothing change 
the windows is fresh and registered 
sorry for using windows it's an ultra small laptop acting as field ground station I will try on linux as soon as I'll back at home
the main problem is that I have no messages no error to guide me ... 
sample video : https://www.youtube.com/watch?v=o-5wfgsqkrE&feature=youtu.be

NaterGator, 2018-01-20:
Can you try to start again while running [this](https://github.com/CobaltFusion/DebugViewPP/releases) as administrator? It will capture all messages passed to the windows debug channel. If you could save and upload the log that would be very helpful.

profff, 2018-01-21:
ok
here's the logs when starting qgc normaly and as admin 
https://drive.google.com/open?id=1_TC1uLr79actKonVJ-fH5jFxL7ynmp34
https://drive.google.com/open?id=1-PyHI6YEMMnY17OU8sl2q4RdYvfMO9Ku
gpu safe and compatibility mode looks the same 
I'm not expert in win system programming (I'm more on real time / firmware side) but it doesn't looks too verbose 
I forgot to say in my first message that I added registry key to got a crash dump but nothing in the qgccrash dir 
thank's for the help

0gopogo, 2018-01-29:
same issue for me on 2 W10 computers, desk and laptop

kyle-irys, 2018-02-17:
Same here - any success?

kyle-irys, 2018-02-17:
I'm having the exact same issues here - no version past 3.0.2 will load.

Windows 10 Home 32-bit 1709 16299.248
QGC 3.2.4 (Emlid Edge Custom Build 2.2, but relevant to anything 3.1.0 and above)
Emlid Edge v1.3

All the logs asked for in the above posts are uploaded here - https://drive.google.com/open?id=1LUsE74bsUk7lBpmSgJY3re5VcCiJGAVH

@NaterGator any support would be greatly appreciated.

NaterGator, 2018-02-17:
Thanks @kyle-irys, having a look now.

NaterGator, 2018-02-17:
Based on the stack I can unwind from the Qt release symbols, it looks like this is related to the FlightDisplayViewUVC component creation:
[image]

Going to dig around, but while I do it would be helpful to know... @kyle-irys, @0gopogo, and @profff: do your computers have an integrated (or connected) camera?

DonLakeFlyer, 2018-02-17:
@NaterGator Thanks a ton for digging into this.

NaterGator, 2018-02-17:
*Sigh.* This is a Qt bug. It stems from the DSCameraSession::updateSourceCapabilities() method:
http://code.qt.io/cgit/qt/qtmultimedia.git/tree/src/plugins/directshow/camera/dscamerasession.cpp?h=v5.9.1#n957

They get an interface instance to an IAMVideoControl on line 971, check that it is valid and use it on lines 974-996. If it was valid they release it on line 995, and then later on line 1028 they check the pointer again and use the interface on line 1036 (use after release).

I'll submit a bug upstream; I'm not sure there's a great fix for this right now, but let me poke around and see what can be done. 👎

DonLakeFlyer, 2018-02-17:
Windows only? Disable UVC on windows? If so, would need to go into Stable_V3.3 branch.

NaterGator, 2018-02-17:
It will definitely be Windows only; this is in the DirectShow plugin for QtMultimedia. That's potentially a viable fix... I'm honestly not sure of the purpose of this. Why are we using UVC in the first place? To support USB analog video decoders?

DonLakeFlyer, 2018-02-17:
I'm not quite sure how it's used myself. All I know is that it does seem to be asked about a lot. @dogmaphobic Would know.

kyle-irys, 2018-02-17:
@NaterGator thanks heaps for your work! 
CONFIRM: The tablet I'm running QGC on has an integrated webcam on it.

I haven't a clue about the rest of what you've said but I am super gad to hear you've found the source somewhat. Am I wrong to suggest it is hardware/compatibility related with regards to the camera question? If so, is there any short term fixes to get it running, such as removing/uninstalling the webcam hardware? Are there also any particular sets of hardware I could avoid when purchasing a different tablet to avoid the issue?

**Waiting** for a stable build with a fix and then the subsequent custom release from Emlid isn't an option.

NaterGator, 2018-02-17:
@kyle-irys can you disable the webcam in the device manager and see if qgc starts? That's the fastest way for you to get going at the moment.

kyle-irys, 2018-02-18:
@NaterGator 
CONFIRMED WORKING
Disabling the camera in Device Manager allowed the program to run. I haven't had the time to work through it and make sure it is actually stable and functioning properly connecting to my vehicle, but have created new logs and uploaded them to the same folder for you to review.

I'll report back when I've played through the program to confirm if it is stable or not.

DonLakeFlyer, 2018-06-20:
Can someone try this again using new builds which are on Qt 5.11? Wondering if Qt fixed it.

Black6spdZ, 2021-04-16:
I am in the same boat, only old version works.. what happened?!

closed 2026-10-04 by github-actions[bot]