mavlink/qgroundcontrol #6058: Crash after downloading ~ 1500 mission items

julianoes, 2018-01-23:
QGC master (e945aa156f6964ae9ae2883e297f151797e7088c) on Fedora Linux crashes after downloading ~1500 mission items.

I found that it crashes in `terrain->queryTerrainData(rgCoord);`
https://github.com/mavlink/qgroundcontrol/blob/master/src/MissionManager/VisualMissionItem.cc#L179. 

The error when crashing is:
```
GLib-ERROR **: Cannot create pipe main loop wake-up: Too many open files The program has unexpectedly finished.
```

From what I can tell is that for each mission item a new `terrain` instance is created with yet another `QNetworkAccessManager` instance. This causes 500+ threads to be spawned and eventually it crashes.

I think the right way to use the terrain class would be to append all mission item coordinates first, and then do one API call, as described in:
https://github.com/mavlink/qgroundcontrol/blob/master/src/Terrain.h#L16-L24

DonLakeFlyer, 2018-01-31:
Hmm, yeah that is some pretty major nastiness.

DonLakeFlyer, 2018-01-31:
@birchera Do you think I should try to fix this or will your cached terrain pull basically make this work decently? I could create my own caching mechanism to stack requests together. But it seems like that just duplicates that the terrain cache will end up doing.

dogmaphobic, 2018-01-31:
There should be one single instance of `QNetworkAccessManager` for these "tiles". That's how the regular map tile engine works. I won't have the time to look into this for another couple of days. If it's still broken by then I can take a look. It's on my todo list to see if I can parse the Json blobs and store the binary data in the database instead any way.

birchera, 2018-01-31:
I am not sure this problem would be fixed, as also with the caching we still have a QNetworkAccessManager per instance of terrain. We just don't use it internally but pass it to the QGeoTiledMapReplyQGC, where I am not exactly sure what happens.
I believe the offline map feature specifies number of concurrent queries, would that limit the number of running threads as well @dogmaphobic?

What could help, is that once we have a certain region in cache, we don't need to query the database any more.

> I think the right way to use the terrain class would be to append all mission item coordinates first, and then do one API call

with the major drawback that you need to update them all together.

DonLakeFlyer, 2018-01-31:
@julianoes If you still have that 1500 mission item plan can you zip and attach here?

julianoes, 2018-02-01:
I don't have a .plan but I think you can just create a fairly big survey, upload it and then download it again.

DonLakeFlyer, 2018-02-01:
I just verified 1800+ download without crashing.

closed 2018-02-01 by DonLakeFlyer

julianoes, 2018-02-02:
Good news, thank you.