mavlink/qgroundcontrol #6062: Request: improved Safety summary pane

Antiheavy, 2018-01-26:
Submitted for consideration:

The "Safety" summary pane in the vehicle setup page doesn't really match the Safety tab settings.  My thought is people want to see a summary of the failsafe behaviors in this pane and can go to the Safety tab if they want to see the full list, details, and options.

Here is a suggested re-work of the Safety summary pane, becoming a Failsafes summary pane instead.  This better matches the order and wording of items in the full Safety tab.  Red text is only to show differences.

[image]

DonLakeFlyer, 2018-05-01:
I wasn't comfortable changing from Safety to Failsafes since it doesn't quite cover fully what is in that page. I think Safety works better. I could be convinced otherwise at some later point.

closed 2018-05-01 by DonLakeFlyer

hamishwillee, 2018-05-01:
> I wasn't comfortable changing from Safety to Failsafes since it doesn't quite cover fully what is in that page. I think Safety works better. I could be convinced otherwise at some later point.

I agree with you. I think most users are more likely to find what they want under the "Safety" name.