mavlink/qgroundcontrol #6077: Unable to upload or download flight plans using a TCP comm link

Jeblas, 2018-02-01:
**QGroundControl Version:** Development HEAD:cdb4cdfc9 2018-01-30 18:42:39 -0800 daily build
 **Operating System:** Windows 10 Enterprise Version: 1607 Build: 14393.1884
**Autopilot Board:** Intel Aero
**Autopilot Firmware:**  PX4

**Steps**
1. Start QGroundControl.
2. Connect to drone via TCP using 0.0.0.0:5760.
3. Create/Load a flight plan.
4. Select Sync -> Upload.

**Error Message:** 
> Mission transfer failed. Retry transfer. Error: Mission write mission count failed, maximum retries exceeded

**Note:** I'm able to download the stored plan using Mission Planner.

DonLakeFlyer, 2018-02-02:
Parameter download across this link is working fine?

Jeblas, 2018-02-06:
Yes, I was able to update parameters without any issues.

DonLakeFlyer, 2018-02-08:
Can you turn on PlanManagerLog as described here: https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html

Try the sequence again. Save the log and attach here.

Jeblas, 2018-02-12:
[2-12 qgc mission download error.txt](https://github.com/mavlink/qgroundcontrol/files/1717997/2-12.qgc.mission.download.error.txt)

DonLakeFlyer, 2018-02-14:
Wow, the firmware is sending some pretty screwed up data. From the log it shows that the vehicle is sending MISSION_ITEM messages to QGC with mission_type set to 7. Two problems there:
* It should be 0, which indicates a regular old mission item
* 7 as a value for MISSION_TYPE doesn't even exist.

Not much QGC can do with that. Mission Planner can download most likely because it does not yet support the new MISSION_ITEM protocol or pay any attention to it.

closed 2024-09-21 by HTRamsey