mavlink/qgroundcontrol #6080: Mission not displayed after initially downloaded

julianoes, 2018-02-01:
Is there a reason that a mission is not displayed in the mission view after it is initially downloaded from the vehicle?

I realize that when you freshly start QGC and connect a Pixhawk with an existing mission, the mission download happens, however the mission is not displayed. It is only displayed once you click Sync -> Download where it actually downloads it again.

DonLakeFlyer, 2018-02-01:
I tries to do that, but there are holes where it gets complicated to make it happen. All depends on the order you do things. Let me look to see if is broken from where it used to be.

DonLakeFlyer, 2018-02-01:
The base case works:
* Boot QGC
* Stay on Fly view
* Connect vehicle
* Go to Plan
* Mission automatically shows up in Plan

But if you do thing in any different order it falls on it's face. Mainly because there are all sorts of user model issues of change the Plan view out from under the user. Then you throw multi-vehicle in there and all hell breaks loose. I tried to make it better a while back. But after a while a backed it all out because if did the wrong thing half the time.

I'll take another look at this in 3.4

julianoes, 2018-02-02:
Ok thx. Not urgent 😄

DonLakeFlyer, 2018-06-18:
Should be working better now

closed 2018-06-18 by DonLakeFlyer