mavlink/qgroundcontrol #6081: AppImage crashes on Fedora 27

julianoes, 2018-02-01:
Trying to start the AppImage on Fedora, I get a crash just after the window appears.

Running the commands in `AppRun` one by one and then debugging using valgrind gives:

```
valgrind "${HERE}/QGroundControl"
==16262== Memcheck, a memory error detector
==16262== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==16262== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==16262== Command: /mnt/appimage/QGroundControl
==16262== 
Settings location "/home/julianoes/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
Failed to set Geoclue positioning requirements. Geoclue error: org.qtproject.QtDBus.Error.InvalidObjectPath
MAVLinkLogManagerLog: MAVLink logs directory: "/home/julianoes/Documents/QGroundControl/Logs"
Map Cache in: "/home/julianoes/.cache/QGCMapCache300" / "qgcMapCache.db"
qml: 0.02962962962962963
qml: _activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()
qml: _activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()
==16262== Use of uninitialised value of size 8
==16262==    at 0x4595DE: __base_ctor  (qcolor.h:281)
==16262==    by 0x4595DE: thumbJoystick (QGCMapPalette.h:57)
==16262==    by 0x4595DE: QGCMapPalette::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (moc_QGCMapPalette.cpp:121)
==16262==    by 0x6DA17A: QGCMapPalette::qt_metacall(QMetaObject::Call, int, void**) (moc_QGCMapPalette.cpp:173)
==16262==    by 0x8E5CB92: ??? (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8E5EC7C: QV4::QQmlValueTypeWrapper::write(QObject*, int) const (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8E57859: QQmlBinding::write(QQmlPropertyData const&, QV4::Value const&, bool, QFlags<QQmlPropertyPrivate::WriteFlag>) (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8E58847: QQmlBinding::update(QFlags<QQmlPropertyPrivate::WriteFlag>) (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8E61CA6: ??? (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8DF5399: ??? (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8DF5DE1: QQmlEnginePrivate::incubate(QQmlIncubator&, QQmlContextData*) (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8DF0D25: QQmlComponent::create(QQmlIncubator&, QQmlContext*, QQmlContext*) (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x6E2A529: ??? (in /mnt/appimage/Qt/libs/libQt5Quick.so.5)
==16262==    by 0x6E2A6A7: ??? (in /mnt/appimage/Qt/libs/libQt5Quick.so.5)
==16262== 
==16262== Conditional jump or move depends on uninitialised value(s)
==16262==    at 0x459555: setLightColors (QGCMapPalette.cc:28)
==16262==    by 0x459555: QGCMapPalette::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (moc_QGCMapPalette.cpp:129)
==16262==    by 0x6DA17A: QGCMapPalette::qt_metacall(QMetaObject::Call, int, void**) (moc_QGCMapPalette.cpp:173)
==16262==    by 0x8DEE201: QQmlPropertyPrivate::write(QObject*, QQmlPropertyData const&, QVariant const&, QQmlContextData*, QFlags<QQmlPropertyPrivate::WriteFlag>) (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8DEF672: QQmlPropertyPrivate::writeValueProperty(QObject*, QQmlPropertyData const&, QVariant const&, QQmlContextData*, QFlags<QQmlPropertyPrivate::WriteFlag>) (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8E56E0F: QQmlBinding::write(QQmlPropertyData const&, QV4::Value const&, bool, QFlags<QQmlPropertyPrivate::WriteFlag>) (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8E58847: QQmlBinding::update(QFlags<QQmlPropertyPrivate::WriteFlag>) (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8E61CA6: ??? (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8DF5399: ??? (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8DF5DE1: QQmlEnginePrivate::incubate(QQmlIncubator&, QQmlContextData*) (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x8DF0D25: QQmlComponent::create(QQmlIncubator&, QQmlContext*, QQmlContext*) (in /mnt/appimage/Qt/libs/libQt5Qml.so.5)
==16262==    by 0x6E2A529: ??? (in /mnt/appimage/Qt/libs/libQt5Quick.so.5)
==16262==    by 0x6E2A6A7: ??? (in /mnt/appimage/Qt/libs/libQt5Quick.so.5)
==16262== 
qml: 34.5 64
qml: _activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()
==16262== Warning: noted but unhandled ioctl 0x6458 with no size/direction hints.
==16262==    This could cause spurious value errors to appear.
==16262==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
Orphaned log file count 0
==16262== Invalid read of size 8
==16262==    at 0x928F49F: ??? (in /mnt/appimage/Qt/libs/libQt5Network.so.5)
==16262==    by 0x92901A3: ??? (in /mnt/appimage/Qt/libs/libQt5Network.so.5)
==16262==    by 0x927CEC4: QSslCertificate::fromData(QByteArray const&, QSsl::EncodingFormat) (in /mnt/appimage/Qt/libs/libQt5Network.so.5)
==16262==    by 0x927EDD1: QSslCertificate::fromPath(QString const&, QSsl::EncodingFormat, QRegExp::PatternSyntax) (in /mnt/appimage/Qt/libs/libQt5Network.so.5)
==16262==    by 0x9299262: ??? (in /mnt/appimage/Qt/libs/libQt5Network.so.5)
==16262==    by 0x92996B3: ??? (in /mnt/appimage/Qt/libs/libQt5Network.so.5)
==16262==    by 0x927CDF4: QSslCertificate::QSslCertificate(QByteArray const&, QSsl::EncodingFormat) (in /mnt/appimage/Qt/libs/libQt5Network.so.5)
==16262==    by 0x927FBE8: QSslConfiguration::QSslConfiguration() (in /mnt/appimage/Qt/libs/libQt5Network.so.5)
==16262==    by 0x9218289: ??? (in /mnt/appimage/Qt/libs/libQt5Network.so.5)
==16262==    by 0x9220281: ??? (in /mnt/appimage/Qt/libs/libQt5Network.so.5)
==16262==    by 0x91FD91F: QNetworkAccessManager::createRequest(QNetworkAccessManager::Operation, QNetworkRequest const&, QIODevice*) (in /mnt/appimage/Qt/libs/libQt5Network.so.5)
==16262==    by 0x91FADF3: QNetworkAccessManager::get(QNetworkRequest const&) (in /mnt/appimage/Qt/libs/libQt5Network.so.5)
==16262==  Address 0x1e08b480 is 8 bytes after a block of size 24 alloc'd
[... 109 lines cut ...]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7fffe0155700 (LWP 17040)]
Settings location "/home/julianoes/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
[New Thread 0x7fffdf30f700 (LWP 17041)]
[New Thread 0x7fffde834700 (LWP 17043)]
Failed to set Geoclue positioning requirements. Geoclue error: org.qtproject.QtDBus.Error.InvalidObjectPath
MAVLinkLogManagerLog: MAVLink logs directory: "/home/julianoes/Documents/QGroundControl/Logs"
Map Cache in: "/home/julianoes/.cache/QGCMapCache300" / "qgcMapCache.db"
[New Thread 0x7fffdde2c700 (LWP 17044)]
[New Thread 0x7fffdd35b700 (LWP 17045)]
[New Thread 0x7fffc7fff700 (LWP 17046)]
[New Thread 0x7fffc5876700 (LWP 17048)]
qml: 0.02962962962962963
qml: _activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()
qml: _activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()
qml: 34.5 64
qml: _activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()
Orphaned log file count 0

Thread 1 "QGroundControl" received signal SIGSEGV, Segmentation fault.
0x00007ffff380a4a3 in ?? () from /mnt/appimage/Qt/libs/libQt5Network.so.5
Missing separate debuginfos, use: dnf debuginfo-install alsa-lib-1.1.5-1.fc27.x86_64 bzip2-libs-1.0.6-24.fc27.x86_64 dbus-libs-1.12.0-1.fc27.x86_64 espeak-1.48.04-12.fc27.x86_64 expat-2.2.5-1.fc27.x86_64 flac-libs-1.3.2-4.fc27.x86_64 fontconfig-2.12.6-4.fc27.x86_64 freetype-2.8-7.fc27.x86_64 glib2-2.54.3-2.fc27.x86_64 gsm-1.0.17-3.fc27.x86_64 gstreamer1-1.12.4-1.fc27.x86_64 gstreamer1-plugins-base-1.12.4-1.fc27.x86_64 libICE-1.0.9-11.fc27.x86_64 libSM-1.2.2-7.fc27.x86_64 libX11-1.6.5-4.fc27.x86_64 libX11-xcb-1.6.5-4.fc27.x86_64 libXScrnSaver-1.2.2-13.fc27.x86_64 libXau-1.0.8-9.fc27.x86_64 libXcursor-1.1.14-10.fc27.x86_64 libXdamage-1.1.4-11.fc27.x86_64 libXext-1.3.3-7.fc27.x86_64 [... cut ...] s-1.8.0-1.fc27.x86_64 mesa-dri-drivers-17.2.4-3.fc27.x86_64 mesa-libGL-17.2.4-3.fc27.x86_64 mesa-libglapi-17.2.4-3.fc27.x86_64 mesa-libwayland-egl-17.2.4-3.fc27.x86_64 nss-mdns-0.10-20.fc27.x86_64 openssl-devel-1.1.0g-1.fc27.x86_64 orc-0.4.28-1.fc27.x86_64 pcre-8.41-4.fc27.x86_64 pcre2-10.30-5.fc27.x86_64 pulseaudio-libs-11.1-7.fc27.x86_64 sssd-client-1.16.0-5.fc27.x86_64 systemd-libs-234-9.fc27.x86_64 tcp_wrappers-libs-7.6-87.fc27.x86_64 xz-libs-5.2.3-4.fc27.x86_64 zlib-1.2.11-4.fc27.x86_64
```

DonLakeFlyer, 2018-02-01:
I think I fixed this. Can you try new daily.? AppImage at least works for me now on Ubuntu 16.04.

julianoes, 2018-02-02:
Tested latest daily. You're right, it works.
Thanks a lot!

closed 2018-02-02 by julianoes

DonLakeFlyer, 2018-02-02:
Phew!